commaai/openpilot #32742: `pkill -f controlsd` doesn't work anymore

adeebshihadeh, 2024-06-13:
Result of #32716

Should we change how we launch Python processes? Or add back `setproctitle`?

https://github.com/commaai/openpilot/blob/6552d4ecb87211b91310ca5d7a7bcb918df36021/system/manager/process.py#L50

schlimeszn, 2024-06-14:
I thought the end goal of the rewrite was to manage a process cleanly. I didnt notice that we already have a process manager. I think we should change how we launch process because then we can give each process custom data fields like its name and PID. Seems like good hooks to have for future development.

adeebshihadeh, 2024-06-14:
Want to make a PR to make the Python processes use the native launcher? I'm assuming it was written that way for two reasons: pre-importing dependencies to speedup launching processes and generic logging for crashes. I suspect #1 is not a problem on C3/3X, but we should test it. #2 can be worked around.

schlimeszn, 2024-06-14:
It can also be because execvp works differently in windows and unix. Saw the launcher code was written 3 years ago, idk if windows was in the development pipeline then. 
https://blog.devgenius.io/execvp-system-call-in-python-everything-you-need-to-know-c402fe6886eb
```Thus, os.execvp behaves differently on Windows and on UNIX systems. On Windows, it spawns or creates a new process (a child or a subprocess) whereas, on a UNIX system, it replaces the original process with a new process!```

schlimeszn, 2024-06-14:
Also, ```pkill -f controlsd``` does work on my MacBook in a container. I ran ```python3 selfdrive/controls/controlsd.py``` and monitored in ```htop```. Also get a error 
```Traceback (most recent call last):
  File "/workspaces/openpilot/selfdrive/controls/controlsd.py", line 848, in <module>
    main()
  File "/workspaces/openpilot/selfdrive/controls/controlsd.py", line 843, in main
    controls = Controls()
               ^^^^^^^^^^
  File "/workspaces/openpilot/selfdrive/controls/controlsd.py", line 67, in __init__
    with car.CarParams.from_bytes(self.params.get("CarParams", block=True)) as msg:
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "common/params_pyx.pyx", line 67, in common.params_pyx.Params.get
    raise KeyboardInterrupt
KeyboardInterrupt
``` 
to confirm. I can still put up a pr to streamline the launch process, but I'm assuming this is a problem when ssh onto a comma device.

schlimeszn, 2024-06-14:
> Also, `pkill -f controlsd` does work on my MacBook in a container. I ran `python3 selfdrive/controls/controlsd.py` and monitored in `htop`. Also get a error

In htop, the name shows as ```python3```, but the pkill command works. I think we are jumping through tooo many naming conventions

schlimeszn, 2024-06-14:
> > Also, `pkill -f controlsd` does work on my MacBook in a container. I ran `python3 selfdrive/controls/controlsd.py` and monitored in `htop`. Also get a error
> 
> In htop, the name shows as `python3`, but the pkill command works. I think we are jumping through tooo many naming conventions

[image]

I lied. The ```python3``` seems to be another bug because it appears when I do ```./selfdrive/modeld/modeld``` also

closed 2024-08-05 by adeebshihadeh