commaai/openpilot #32761: Enforce using `python3` instead of `python`

adeebshihadeh, 2024-06-16:
I'm trying out 24.04 where we want to be able to use the system python, but I'm hitting this:
```
adeeb:~$ python
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3
```

What do other projects do?

BBBmau, 2024-06-16:
Looks like they opted out of including use/bin/python in 24.04

https://utcc.utoronto.ca/~cks/space/blog/python/Ubuntu2404PythonState

We could include a step in the setup to do some kind of aliasing

adeebshihadeh, 2024-06-16:
That's not the problem here. AFAIK `python` is always supposed to be python2 (or at least ambiguous) and `python3` should be preferred. We used pyenv to make `python` always refer to our python install, however we want to drop the need for pyenv soon.

AbhishekChaudharii, 2024-06-21:
Installing python-is-python3 package might solve your issue.
`sudo apt install python-is-python3`

The package includes a symbolic link that directs the `/usr/bin/python` interpreter to the current default python3 version.

closed 2024-08-14 by maxime-desroches

adeebshihadeh, 2024-09-04:
@maxime-desroches is this done?

maxime-desroches, 2024-09-04:
@adeebshihadeh Yes

adeebshihadeh, 2024-09-04:
Oh, weird still shows as open in the project view for some reason.

maxime-desroches, 2024-09-04:
Yes, weird bug. I've tried many times to make it purple