commaai/openpilot #32762: waiting for controls to start after update to 0.9.7

jyoung8607, 2024-06-16:
### Describe the bug

After this user received an automatic upgrade from 0.9.6-release to 0.9.7-release, they experienced "Waiting for controls to start" each time they tried to drive.

Tracebacks in the logs show core affinity couldn't be set for some critical processes, which implies one or more of the 845 performance cores might not have come online when transitioning offroad->onroad.

Ignition/onroad cycles did not clear the problem, but a power cycle did clear the problem.

### Provide a route where the issue occurs

b08418a961021457/00000001--8dec252c1f

### openpilot version

0.9.7

### Additional info

```
[0.005306] system/loggerd/loggerd.cc:218 - loggerd_thread: logging accelerometer
[0.006482] process.py:291 - : starting python selfdrive.monitoring.dmonitoringd
[0.006877] process.py:46 - : crash
Traceback (most recent call last):
  File "/data/openpilot/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 844, in main
    config_realtime_process(4, Priority.CTRL_HIGH)
  File "/data/openpilot/openpilot/common/realtime.py", line 44, in config_realtime_process
    set_core_affinity(c)
  File "/data/openpilot/openpilot/common/realtime.py", line 37, in set_core_affinity
    os.sched_setaffinity(0, cores)
OSError: [Errno 22] Invalid argument
[0.006926] process.py:46 - : crash
Traceback (most recent call last):
  File "/data/openpilot/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 844, in main
    config_realtime_process(4, Priority.CTRL_HIGH)
  File "/data/openpilot/openpilot/common/realtime.py", line 44, in config_realtime_process
    set_core_affinity(c)
  File "/data/openpilot/openpilot/common/realtime.py", line 37, in set_core_affinity
    os.sched_setaffinity(0, cores)
OSError: [Errno 22] Invalid argument
[0.006993] system/loggerd/loggerd.cc:218 - loggerd_thread: logging accelerometer2
[0.007046] system/loggerd/loggerd.cc:218 - loggerd_thread: logging androidLog
[0.007093] system/loggerd/loggerd.cc:218 - loggerd_thread: logging cameraOdometry
```

jackking444, 2024-08-20:
I get the same issue
Traceback (most recent call last):
  File "/data/openpilot/selfdrive/modeld/modeld.py", line 419, in <module>
    main(demo=args.demo)
  File "/data/openpilot/selfdrive/modeld/modeld.py", line 174, in main
    config_realtime_process(7, 54)
  File "/data/openpilot/openpilot/common/realtime.py", line 46, in config_realtime_process
    set_core_affinity(c)
  File "/data/openpilot/openpilot/common/realtime.py", line 39, in set_core_affinity
    os.sched_setaffinity(0, cores)
OSError: [Errno 22] Invalid argument

jollycar, 2025-01-18:
I have the same problem. If I manage to get logging, I will add it here

jollycar, 2025-01-18:
I have no clue how to get logging of my device. I have looked in the comma.ai site in the FAQ and WIKI, I have looked in Discord but I have tried my best to find info about getting logs off the device. No pinned messages, did not find channels for debugging/etc.

adeebshihadeh, 2025-01-23:
rlogs are gone for that route, and there's nothing good in the bootlog. Gonna close until it pops up again.

closed 2025-01-23 by adeebshihadeh

jyoung8607, 2025-06-04:
> The dmonitoringmodeld.py is hard-coded to 7 cores. The comma3x only has 4 cores.

It has eight. If it's offroad you'll only see four in top/htop because the [perf cores get offlined](https://github.com/commaai/openpilot/blob/2e41d959ac8610f2a91d28627c0f1357b1089453/system/hardware/tici/hardware.py#L365-L368) for power savings.

Edit: sorry I broke your earlier comment, hit the wrong button to reply.

tonesto7, 2025-06-04:
`nproc` returns 4

tonesto7, 2025-06-04:
@jyoung8607 Thanks for the reply!  This points me in a direction to debug.