commaai/openpilot #32767: Simulation Process not running: navd

dennisklad, 2024-06-17:
### Describe the bug

I am creating a Unity plugin to integrate OpenPilot to allow for better simulations and synced my fork with the master.
Now the bridge is not working, because `Process not running: navd`.

I run `export BLOCK=mapsd` in the console, as instructed by the README, but I still get the same error when trying to engage.

Does anybody have a same issue? 

[image]

### OS Version

Ubuntu 20.04

dennisklad, 2024-06-17:
This is the error from the `./launch_openpilot.sh`.

```
Waiting for CAN messages...
logcatd proclogd modeld ui soundd locationd calibrationd torqued controlsd card deleter navd paramsd plannerd radard hardwared statsd
crash
Traceback (most recent call last):
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 361, in main
    route_engine = RouteEngine(sm, pm)
                   ^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 58, in __init__
    self.api = Api(self.params.get("DongleId", encoding='utf8'))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/common/api/__init__.py", line 13, in __init__
    with open(Paths.persist_root()+'/comma/id_rsa') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/n/.comma/persist/comma/id_rsa'
Process navd:
Traceback (most recent call last):
  File "/home/n/.pyenv/versions/3.12.4/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/n/.pyenv/versions/3.12.4/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 361, in main
    route_engine = RouteEngine(sm, pm)
                   ^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 58, in __init__
    self.api = Api(self.params.get("DongleId", encoding='utf8'))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/common/api/__init__.py", line 13, in __init__
    with open(Paths.persist_root()+'/comma/id_rsa') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/n/.comma/persist/comma/id_rsa'

```

dennisklad, 2024-06-18:
After restarting my PC it stopped working again and doing `export BLOCK=mapsd` does not help.

```
Process navd:
Traceback (most recent call last):
  File "/home/n/.pyenv/versions/3.12.4/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/n/.pyenv/versions/3.12.4/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 361, in main
    route_engine = RouteEngine(sm, pm)
                   ^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/selfdrive/navd/navd.py", line 58, in __init__
    self.api = Api(self.params.get("DongleId", encoding='utf8'))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/n/Masterarbeit/openpilot-plugin/openpilot/common/api/__init__.py", line 13, in __init__
    with open(Paths.persist_root()+'/comma/id_rsa') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/n/.comma/persist/comma/id_rsa'
```

MdAkramKhanJehad, 2024-06-18:
> After restarting my PC it stopped working again and doing `export BLOCK=mapsd` does not help.
>
> ```
> Process navd:
> Traceback (most recent call last):
>   File [...]

Have you fixed it already? facing the same issue

sshane, 2024-06-19:
You need to specify `MAPBOX_TOKEN` as an environment variable to get this working, if you don't the Api class assumes this is an official device and expects a private key to our validate into our backend to exist.

Make sure to read the README! https://github.com/commaai/openpilot/blob/master/selfdrive/navd/README.md

closed 2024-06-19 by sshane

dennisklad, 2024-06-19:
Thank you for the reply.
The README in the sim dir mentions that one could also disable navd., but
this doss not work for me.

https://github.com/commaai/openpilot/blob/master/tools%2Fsim%2FREADME.md

dennisklad, 2024-06-22:
@sshane the link to the README you provided does not work.

sshane, 2024-06-22:
Nav was just removed