commaai/openpilot #32771: Turn Signals and Lane Changes

dennisklad, 2024-06-17:
### Describe the bug

The `_run` function in the `tools/sim/bridge/common.py` has some control message types already defined for impementing turns. I have tried to use this messages, but the car does not seem to react at all to them in the virtual environment. Do the lane change and the `blinker_left`/ `blinker_right` messages work or is this still under developement?

```
message = q.get()
[...]
elif m[0] == "blinker":
  if m[1] == "left":
    self.simulator_state.left_blinker = True
  elif m[1] == "right":
    self.simulator_state.right_blinker = True
```

### OS Version

Ubuntu 20.04

dennisklad, 2024-07-01:
Are there any plans to integrate this messages to the MetaDrive simulator so that we can change lanes and potentially decide where to turn in a T-intersection?

adeebshihadeh, 2024-08-09:
This isn't supported yet. Join the simulation channel in Discord for further discussion.

closed 2024-08-09 by adeebshihadeh