commaai/openpilot #32776: no follow distance change on button press

samcooler, 2024-06-18:
### Describe the bug

I have a comma 3, latest stock release 0.9.7, Prius 2016 w/ sdsu.
Follow distance button on the steering wheel has no apparent effect. Settings are not changed, car UI shows nothing.

### Provide a route where the issue occurs

4617eff9ed63ea3f

### openpilot version

0.9.7

adeebshihadeh, 2024-06-18:
We're removing SDSU support from stock openpilot: #32777.

closed 2024-06-18 by adeebshihadeh

samcooler, 2024-06-18:
okay. Not a big deal on that feature, but sounds like I might continue to lose out on some future features. Hopefully someone maintains a fork (plz post below or dm me if you know of it)

@adeebshihadeh  I recommend adding this note to the blog post change log, for future others convenience

adeebshihadeh, 2024-06-18:
We'll note it in the 0.9.8 blog.

jpritikin, 2024-06-18:
I have similar (or same) issue with 2022 Toyota Prius.

FrogAi, 2024-06-18:
@samcooler I'll be maintaining SDSU fork on FrogPilot. Repo is in my profile.

samcooler, 2024-06-18:
@adeebshihadeh curious how this is already occurring in my car which is stock auto-updated to 0.9.6. The follow distance feature is not enabled because of the SDSU but support is not yet dropped? Can you clarify? Should I expect it to work on 0.9.6?

adeebshihadeh, 2024-06-18:
> @adeebshihadeh curious how this is already occurring in my car which is stock auto-updated to 0.9.7. The follow distance feature is not enabled because of the SDSU but support is not yet dropped? Can you clarify? Should I expect it to work on 0.9.7?

Correct that the removal is only in 0.9.8. I don’t know whether it’s expected to work on 0.9.7.