commaai/openpilot #32778: UI doesn't show camera feed for CTF replay route

Quantizr, 2024-06-18:
### Describe the bug

For the route in [CTF](https://github.com/commaai/openpilot/blob/master/tools/CTF.md):
```bash
# start the route reply
cd tools/replay
./replay '0c7f0c7f0c7f0c7f|2021-10-13--13-00-00' --dcam --ecam

# start the UI in another terminal
selfdrive/ui/ui
```

The UI does not switch to the onroad display. The UI is receiving data as the camera feed can be viewed in Settings -> Device -> Driver Camera, the issue is probably with the format of the CTF route itself.

Starting the demo replay (`/replay --demo --dcam --ecam`), stopping the demo, then starting the CTF replay allows the CTF replay to work.

### OS Version

Ubuntu 20.04

sshane, 2024-06-19:
This is an old route that uses the deprecated `pandaState` service for ignition, @adeebshihadeh should we deprecate the CTF or support the old service in UI?

adeebshihadeh, 2024-06-19:
I'd prefer regenerating the route. I think it should be easy: `xx/projects/ctf`.

bravochar, 2024-07-12:
I am running into this issue as well. I can also verify that the driver camera preview will work, but my UI does not switch to the onroad display, even after pausing/resuming the replay or even quitting and reloading the replay.

MikeBusuttil, 2024-12-11:
yo @adeebshihadeh wanna pair on this? 😄

royjr, 2024-12-11:
same here, demo route works fine though

jyoung8607, 2024-12-11:
For benefit of others looking at this issue, Shane and Adeeb already triaged it above, and they're going to fix it by regenerating the CTF route itself. The old CTF route has the ignition-on state in a place we no longer look for it.

robbiewu008, 2025-03-07:
Same issue with MacOS 15.3.1, Any update?

Sanaznor71, 2025-05-01:
Having same problem!
Would be great if you have any update.
Thanks

PrabodhGyawali, 2025-05-23:
I assume we can checkout to 3 years ago when it worked do you know which commit hash to checkout to?

MikeBusuttil, 2025-05-24:
Nah its not the code, its the route that's stored on comma's server.  Nag @sshane or @adeebshihadeh and they'll be able to regenerate that route and fix it.  I hope they do

ealderman-25, 2025-06-20:
Ran into this today.
In the meantime, you can instead run `selfdrive/debug/uiview.py` as a workaround.

MikeBusuttil, 2025-08-29:
nah you can still do the CTF it just makes it a little trickier to get started