commaai/openpilot #32781: Navigation pane not loading offroad or onroad

MarcoTheDingo, 2024-06-18:
### Describe the bug

The navigation widget/pane is completely black on the offroad home screen after the June 18th master-ci build. When going onroad, tapping the screen just cycles between the status bar and full screen, no navigation 50/50 split.
Tried re-installing master-ci, same issue.
[image]

### Provide a route where the issue occurs

5130484aa8069bad | No route needed

### openpilot version

0.9.8 | master-ci | e4fcf42ada2206b8cd4d5c9927d8792c7701faa8

eFiniLan, 2024-06-18:
@MarcoTheDingo Relevent:

https://github.com/commaai/openpilot/pull/32773/files

https://github.com/commaai/openpilot/blob/5c4ea14a3cb7bcb443fbee8c5b1d52c10788c9e8/selfdrive/ui/SConscript#L11

adeebshihadeh, 2024-06-20:
#32773

closed 2024-06-20 by adeebshihadeh