commaai/openpilot #32791: [$100 bounty] Bring back macOS CI

adeebshihadeh, 2024-06-19:
We had macOS CI for a long time but it was hard to maintain, in part due to the package cache breaking all the time. Now that dependencies are down, let's see how fast it is without a package cache.

Just run `tools/mac_setup.sh` and `scons -h` after to make sure it vaguely worked. If that's fast enough to leave enabled on master, we can slowly bring back compiling openpilot, running tests, etc.

adeebshihadeh, 2024-06-19:
I'm thinking just Apple Silicon.

closed 2024-06-26 by maxime-desroches

leomonde, 2024-06-26:
Sorry, I don't understand, issue is closed but after git pull I still get error compiling (scons -u -j$(nproc)) in OSX 14.5 or Ubuntu 22.04 or Ubuntu 24.04. Did I miss something?
My CPU is Apple Silicon (ARM M2), I run Ubuntu over Parallel VM. Thanks.

adeebshihadeh, 2024-06-26:
macOS setup is run, but it doesn't build yet

leomonde, 2024-06-26:
> macOS setup is run, but it doesn't build yet

Is there a way to build at least tools?
With Ubuntu in VM, neither 22 or 24, it doesn't even finish install. Is my CPU the problem?

PR #32819 (2024-06-26): Bring back macOS CI
Fixes #32791
Runs tools/mac_setup.sh and scons -h without any caching. mac_setup.sh takes ~4 minutes.