commaai/openpilot #32809: ubuntu_setup.sh failure installing casadi (Ubuntu 24.04 and ARM M2 CPU)

leomonde, 2024-06-21:
### Describe the bug

Installing casadi (3.6.5) Failed
ChefBuildError
Backend subprocess exited when trying to invoke build_wheel

### OS Version

Ubuntu 24.04 LTS

leomonde, 2024-06-21:
Ubuntu 24.04 running in VM environment (Parallel)

maxime-desroches, 2024-06-21:
We dropped support for arm macOS during the transition to python3.12/ubuntu24.04 and one of the reasons was casadi having problems building. An issue (https://github.com/commaai/openpilot/issues/32791) is open to bring it back.

leomonde, 2024-06-21:
> We dropped support for arm macOS during the transition to python3.12/ubuntu24.04 and one of the reasons was casadi having problems building. An issue (https://github.com/commaai/openpilot/issues/32791) is open to bring it back.

I am not using MacOS.
I am using Ubuntu 24.04!

athaarvv, 2024-06-24:
is there any resolution found for casadi issue ?

leomonde, 2024-06-24:
> is there any resolution found for casadi issue ?

I don't know.
I tried to go back to commit cb63f10 (git reset), to use an old OP version in Ubuntu 22 but it didn't work.
Strange... as it used to work months ago.

adeebshihadeh, 2024-08-09:
New casadi release is out which should fix this.

closed 2024-08-09 by adeebshihadeh