commaai/openpilot #32812: New docs.comma.ai

adeebshihadeh, 2024-06-23:
https://docs.comma.ai is currently just a nicer way to view all the READMEs in the repo.  https://rerun.io/docs are the nicest docs I've seen recently. They look great and the content flows nicely. I'm thinking we'll want something like this:
* copy the rerun "Getting Started" section: what is openpilot, quick start, installing, etc.
* walkthroughs:
  * "how to replay a route"
  * "how to debug a process crash"
  * "how to live stream the cameras to your PC"
* car porting section
  * brand port
  * model port

ugtthis, 2024-06-23:
[docs.howtocomma.com](https://docs.howtocomma.com/docs/what-is-openpilot) is my attempt of a better way to do docs geared towards new users who know very little about openpilot and how to install the device. Maybe we can cherry pick any ideas from my attempt to then implement in the new docs. Lots of the content from docs.howtocomma was from the community GitHub. The biggest difference from the community GitHub docs and howtocomma docs is the structure and flow of information

ugtthis, 2024-06-23:
Something that would get more newer developers to try to contribute to openpilot(if this was a goal) would be to have a hello world type of step by step document for openpilot. The closest one to that is the [Turning the speed blue](https://blog.comma.ai/turning-the-speed-blue/) blog post which is using the terminal versus some sort of IDE, and also doesn't help get new people more familiar with the code base.

My current favorite example of a document that really holds your hand and walks you through what different folders mean, code examples w/explanations, and such are the [expo docs](https://docs.expo.dev/tutorial/introduction/)

adeebshihadeh, 2024-06-23:
This is for developer documentation. I started a new discussion in #current-projects on Discord, let's continue discussing there.

adeebshihadeh, 2024-06-24:
$250 bounty to get an MVP of this shipped to docs.comma.ai. Concretely, that's:
* enough content to be more useful than current docs (should mostly be moving existing README's)
* CI pushes to `commaai/openpilot-docs` (replacing the existing docs stuff). this handles deploying to https://docs.comma.ai
* maybe have some new people try it out on Discord before we switch?

The key is that we don't need a lot of content to start. The important thing is getting the structure and flow setup properly.

closed 2024-07-28 by adeebshihadeh