commaai/openpilot #32878: Cython build error, due to bad cache?

adeebshihadeh, 2024-06-30:
Got this error on my home PC after pulling and running ubuntu_setup.sh after a few weeks. Not sure exactly what commit it was on before that.

Looks like there's some dependency not setup properly in our SConscripts. @maxime-desroches is this the same as what you were seeing in CI?

[build_error.txt](https://github.com/user-attachments/files/16045388/build_error.txt)

```
common/transformations/transformations.cpp:3841:13: error: use of undeclared identifier 'PyDataType_FLAGS'
 3841 |   __pyx_r = PyDataType_FLAGS(__pyx_v_self);
      |             ^
common/transformations/transformations.cpp:3875:13: error: use of undeclared identifier 'PyArray_MultiIter_NUMITER'
 3875 |   __pyx_r = PyArray_MultiIter_NUMITER(__pyx_v_self);
      |             ^
common/transformations/transformations.cpp:3909:13: error: use of undeclared identifier 'PyArray_MultiIter_SIZE'; did you mean 'PyArrayMultiIterObject'?
 3909 |   __pyx_r = PyArray_MultiIter_SIZE(__pyx_v_self);
      |             ^
.venv/lib/python3.12/site-packages/numpy/core/include/numpy/ndarraytypes.h:1263:3: note: 'PyArrayMultiIterObject' declared here
 1263 | } PyArrayMultiIterObject;
      |   ^
common/transformations/transformations.cpp:3943:13: error: use of undeclared identifier 'PyArray_MultiIter_INDEX'
 3943 |   __pyx_r = PyArray_MultiIter_INDEX(__pyx_v_self);
      |             ^
common/transformations/transformations.cpp:3977:13: error: use of undeclared identifier 'PyArray_MultiIter_NDIM'
 3977 |   __pyx_r = PyArray_MultiIter_NDIM(__pyx_v_self);
      |             ^
common/transformations/transformations.cpp:4011:13: error: use of undeclared identifier 'PyArray_MultiIter_DIMS'
 4011 |   __pyx_r = PyArray_MultiIter_DIMS(__pyx_v_self);
      |             ^
common/transformations/transformations.cpp:4045:13: error: use of undeclared identifier 'PyArray_MultiIter_ITERS'; did you mean 'PyArrayMultiIterObject'?
 4045 |   __pyx_r = PyArray_MultiIter_ITERS(__pyx_v_self);
      |             ^
.venv/lib/python3.12/site-packages/numpy/core/include/numpy/ndarraytypes.h:1263:3: note: 'PyArrayMultiIterObject' declared here
 1263 | } PyArrayMultiIterObject;
      |   ^
12 errors generated.
scons: *** [msgq_repo/msgq/visionipc/visionipc_pyx.o] Error 1
```

adeebshihadeh, 2024-06-30:
Needed to do a full clean to recover.

maxime-desroches, 2024-06-30:
Yes, same type of errors as https://github.com/commaai/openpilot/actions/runs/9670296819/job/26679640212#step:11:283

0312birdzhang, 2024-07-05:
Downgrade numpy to 1.x

adeebshihadeh, 2024-08-09:
@maxime-desroches is this fixed?

maxime-desroches, 2024-08-09:
Yes, the cython builds are now dependent on the numpy version

closed 2024-08-09 by maxime-desroches