commaai/openpilot #32882: Bring back linux aarch64 in CI

maxime-desroches, 2024-07-01:
We recently dropped aarch64 from CI (https://github.com/commaai/openpilot/pull/32715) during the upgrade to ubuntu24.04/python3.12. One of the main reason for this was Casadi not building. See this comment for details: https://github.com/commaai/openpilot/pull/32708#discussion_r1635604873

The goal:
- install all python/ubuntu dependencies and build openpilot on a linux aarch64 machine. (You don't need an aarch64 linux machine for doing this bounty since everything will be validated by our CI runner.)

How to do it:
- add back aarch64 in the build matrix such that it runs on our namespace runner (see [here](https://github.com/commaai/openpilot/pull/32715/files#diff-38aca91846a892e562f5343cde335a052609352cd50edb16d2905db04944a47eL76) for how it was done)
- add back the docker_push_multiarch step (see [here](https://github.com/commaai/openpilot/pull/32715/files#diff-38aca91846a892e562f5343cde335a052609352cd50edb16d2905db04944a47eL93) )
-  make the test pass

0312birdzhang, 2024-07-04:
A stupid but works way is install `python3.12-dev` and `cp /usr/include/python3.12/patchlevel.h /data/openpilot/.venv/include/`.

leomonde, 2024-07-05:
I noted switching to Python 3.11, I am able to install manually casadi 3.6.5 and get OPENPILOT SETUP DONE. 
System is Ubuntu 24.04 ARM (M2 Apple Silicon).
Of course I have to switch back to Python 3.12 right after casadi installation.
(used update-alternatives)

coder351, 2024-07-11:
@maxime-desroches The pypi registry for Casadi does not provide wheel for python 3.12. I have built the wheel for Casadi for aarch64 using python 3.12. I have PR (https://github.com/commaai/openpilot/pull/32968) ready for review. I have added back aarch64 build matrix and the docker_push_multiarch step, but need some help with the build_release step of selfdrive_test.yaml. The build_devel.sh is calling master_ci which is probably configured on the master branch.

leomonde, 2024-07-13:
> @maxime-desroches The pypi registry for Casadi does not provide wheel for python 3.12. I have built the wheel for Casadi for aarch64 using python 3.12. I have PR (#32968) ready for review. I have [...]

Man... you are really "the man"!!!
I was unable to run Cabana and Polotjuggler since months ago because of my damn M2 CPU.
Tried directly on OSX, not working, tried with docker, not working, tried with Ubuntu 24, Ubuntu 22 and Ubuntu 20, not working, tried VM with Parallel and VM with UTM, not working...

I just git clone your fork now... and it installed and compiled without errors directly on OSX! Amazing! Thank you A LOT!
I am just getting some segmentation fault on Cabana, is it normal? Can I do something to avoid? Thanks again!

coder351, 2024-07-17:
@maxime-desroches I have outlined alternate approach in PR which should not require wheel to be pushed to repo or to pypi.

closed 2024-07-22 by maxime-desroches