commaai/openpilot #32902: CI: improve `CI_weekly_report` generation

BBBmau, 2024-07-04:
continuation of #32869 should match the criteria that were set there:

> Suggestions for improving the report:
> 
>     * replace counts with pass/fail percentages, plus total runs on top
> 
>     * color the background of each "Job" cell with the pass percent, gradient from green to red
> 
>     * the actions links in the table aren't links
> 
>     * maybe also replace the text for the links with "here" or something short so the table doesn't change size when opening the dropdown

signed-long, 2024-07-05:
Cell colours is out since github markdown doesn't allow CSS. I think Emojis are the best we got. 

Could do something like this with a jenkins style yellow=flaky: 
# CI Job Summary - 5 Runs
| Job | ✅ Passing | ❌ Failure Details |
| --- | --- | --- |
| 🟢 Create UI Report | 100% |  |
| 🟢 PR comments | 0% |  |
| 🟢 build (x86_64) | 100% |  |
| 🟢 build macos | 100% |  |
| 🟡 build release | 0% | [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028966813), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028968828), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028971259), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028973286), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028975361) |
| 🟢 cars (0) | 100% |  |
| 🟢 cars (1) | 100% |  |
| 🟢 devcontainer | 100% |  |
| 🟢 process replay | 100% |  |
| 🟢 simulator driving | 100% |  |
| 🟢 static analysis | 100% |  |
| 🟢 test 20.04 + 3.11 | 100% |  |
| 🟡 unit tests | 20% | [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028969645), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028972471), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028974090), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028976021) |

Or an emoji gradient: 
# CI Job Summary - 5 Runs
|  |Job | ✅ Passing | ❌ Failure Details |
| --- | --- | --- | --- |
| 🟩🟩🟩🟩🟩| Create UI Report | 100% |  |
| 🟩🟩🟩🟩🟩 | PR comments | 0% |  |
| 🟩🟩🟩🟩🟩 | build (x86_64) | 100% |  |
| 🟩🟩🟩🟩🟩 | build macos | 100% |  |
| 🟥🟥🟥🟥🟥 | build release | 0% | [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028966813), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028968828), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028971259), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028973286), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028975361) |
| 🟩🟩🟩🟩🟩  | cars (0) | 100% |  |
| 🟩🟩🟩🟩🟩  | cars (1) | 100% |  |
| 🟩🟩🟩🟩🟩  | devcontainer | 100% |  |
| 🟩🟩🟩🟩🟩  | process replay | 100% |  |
| 🟩🟩🟩🟩🟩  | simulator driving | 100% |  |
| 🟩🟩🟩🟩🟩  | static analysis | 100% |  |
| 🟩🟩🟩🟩🟩  | test 20.04 + 3.11 | 100% |  |
| 🟩🟥🟥🟥🟥| unit tests | 20% | [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028969645), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028972471), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028974090), [log](https://github.com/signed-long/openpilot/actions/runs/9789363541/job/27028976021) |

adeebshihadeh, 2024-07-05:
Second one looks great! Want to open a PR @signed-long?

PR #32911 (2024-07-09): Update ci weekly report summary

Closes #32902

https://github.com/signed-long/openpilot/actions/runs/9803187594/attempts/1#summary-27069501960
closed 2024-07-06 by maxime-desroches