commaai/openpilot #32941: [$100 bounty] Switch to pytest

maxime-desroches, 2024-07-09:
This is a central issue tracking the switch from unittest to pytest.

The goal:
- Switch to pytest
- Make the tests easy to run

How to do it:
- Convert from unittest to pytest with a tool like https://github.com/dannysepler/pytestify
- Write a good default config in each pyproject.toml such that all the tests can be run by only invoking `pytest`
 
Needs to be done in each of the following repos:
- [ ] cereal
- [ ] panda
- [ ] opendbc
- [ ] rednose
- [ ] teleoprtc

ranjanmangla1, 2024-07-11:
is it still open?

maxime-desroches, 2024-07-11:
@ranjanmangla1 Yes, only panda is left if you want to do it

UkuLoskit, 2024-07-11:
i'm working on panda still

adeebshihadeh, 2024-07-11:
Locked to @UkuLoskit

sudoevans, 2024-08-20:
Seems like the issues wasn't fixed, Did @UkuLoskit make any update?

Pious1020, 2024-09-14:
hey, is the bounty still open for this?

closed 2025-03-16 by adeebshihadeh