commaai/openpilot #32954: Build failure on macOS

adeebshihadeh, 2024-07-09:
Ran `tools/mac_setup.sh` on commit 133f25eecbd7be2343624c8ed3c426663bd027e5

```
(.venv) macbookair:openpilot adeebshihadeh$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Retrieved `selfdrive/ui/translations/main_en.qm' from cache
Retrieved `selfdrive/ui/translations/main_de.qm' from cache
Retrieved `selfdrive/ui/translations/main_fr.qm' from cache
Retrieved `selfdrive/ui/translations/main_pt-BR.qm' from cache
Retrieved `selfdrive/ui/translations/main_es.qm' from cache
Retrieved `selfdrive/ui/translations/main_tr.qm' from cache
Retrieved `selfdrive/ui/translations/main_ar.qm' from cache
Retrieved `selfdrive/ui/translations/main_th.qm' from cache
Retrieved `selfdrive/ui/translations/main_zh-CHT.qm' from cache
Retrieved `selfdrive/ui/translations/main_zh-CHS.qm' from cache
Retrieved `selfdrive/ui/translations/main_ko.qm' from cache
Retrieved `selfdrive/ui/translations/main_ja.qm' from cache
python3 /Users/adeebshihadeh/openpilot/selfdrive/modeld/get_model_metadata.py /Users/adeebshihadeh/openpilot/selfdrive/modeld/models/supercombo.onnx
Retrieved `selfdrive/assets/assets.cc' from cache
clang++ -o selfdrive/modeld/runners/runmodel_pyx.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DTRANSFORM_PATH=\"/Users/adeebshihadeh/openpilot/selfdrive/modeld/transforms/transform.cl\" -DLOADYUV_PATH=\"/Users/adeebshihadeh/openpilot/selfdrive/modeld/transforms/loadyuv.cl\" -g -fPIC -O2 -Wunused -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -Wno-vla-cxx-extension -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I/opt/homebrew/include -I/opt/homebrew/opt/openssl@3.0/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/qrcode -Ithird_party -Icereal -Imsgq -Iopendbc/can -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -I.venv/lib/python3.12/site-packages/numpy/core/include selfdrive/modeld/runners/runmodel_pyx.cpp
clang++ -o selfdrive/modeld/runners/snpemodel_pyx.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DTRANSFORM_PATH=\"/Users/adeebshihadeh/openpilot/selfdrive/modeld/transforms/transform.cl\" -DLOADYUV_PATH=\"/Users/adeebshihadeh/openpilot/selfdrive/modeld/transforms/loadyuv.cl\" -g -fPIC -O2 -Wunused -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -Wno-vla-cxx-extension -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I/opt/homebrew/include -I/opt/homebrew/opt/openssl@3.0/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/qrcode -Ithird_party -Icereal -Imsgq -Iopendbc/can -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -I.venv/lib/python3.12/site-packages/numpy/core/include selfdrive/modeld/runners/snpemodel_pyx.cpp
clang++ -o selfdrive/pandad/can_list_to_can_capnp.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -Wno-vla-cxx-extension -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -I/opt/homebrew/include -I/opt/homebrew/opt/openssl@3.0/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/qrcode -Ithird_party -Icereal -Imsgq -Iopendbc/can -Ithird_party/json11 selfdrive/pandad/can_list_to_can_capnp.cc
Traceback (most recent call last):
  File "/Users/adeebshihadeh/openpilot/selfdrive/modeld/get_model_metadata.py", line 15, in <module>
    model = onnx.load(str(model_path))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/adeebshihadeh/openpilot/.venv/lib/python3.12/site-packages/onnx/__init__.py", line 210, in load_model
    model = _get_serializer(format, f).deserialize_proto(_load_bytes(f), ModelProto())
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/adeebshihadeh/openpilot/.venv/lib/python3.12/site-packages/onnx/serialization.py", line 118, in deserialize_proto
    decoded = typing.cast(Optional[int], proto.ParseFromString(serialized))
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
google.protobuf.message.DecodeError: Error parsing message
scons: *** [selfdrive/modeld/models/supercombo_metadata.pkl] Error 1
scons: building terminated because of errors.

```

eFiniLan, 2024-07-10:
I had the same issue with PC version.
my environment is on a docker version of ubuntu 20.04, I reset the environment and moved ubuntu 24.04.

If I do a fresh clone and build from it and it worked.

I wonder if it's related to something that cached inside the openpilot/ (scons --clean didnt manage to remove that)

eFiniLan, 2024-07-19:
I just noticed that the supercombo.onnx file did not download properly. Run the following commands:
```
git lfs fetch
git lfs update
```
This should resolve the issue. I suspect this happens when switching between branches that contain different models.

adeebshihadeh, 2024-07-19:
I forgot I was having LFS issues on this machine
```
macbookair:openpilot adeebshihadeh$ git lfs pull
batch request: remote: :   0% (0/1), 0 B | 0 B/s                               
remote: ========================================================================
remote: 
remote: ERROR: The namespace you were looking for could not be found.

remote: 
remote: ========================================================================
remote:: exit status 1
Failed to fetch some objects from 'https://gitlab.com/commaai/openpilot-lfs.git/info/lfs'
```

closed 2024-07-19 by adeebshihadeh