commaai/openpilot #32999: Mismatch in allowable steering angle in Openpilot and panda for Tesla: Openpilot clip angle to current angle of the car to avoid EPS fault

bongbui321, 2024-07-15:
### Describe the bug

https://github.com/commaai/openpilot/blob/aa8484093e886cd8ae90e4eb42118306794fc439/selfdrive/car/tesla/carcontroller.py#L34

This line of code clips the steering angle to within allowable range of current steering angle of a car. This creates a mismatch in logic between openpilot and panda. Panda only calculate the allowable steering angle from the vehicle speed min and last desired angle

I assume the scenario would happen when openpilot first tries to engage when wheels are at a large enough angle

### Provide a route where the issue occurs

N/A

### openpilot version

latest commit - #32948

sshane, 2024-07-17:
Perhaps we should just move this up a few lines so the rate limiting is done after this limit? Then it should match

sshane, 2024-07-18:
Maybe @robbederks can expand on this exact limit and its purpose.

robbederks, 2024-07-18:
Has never caused me issues in the field (and the epas faults immediately when the delta between commanded and actual angle is too high), but the logic should probably match between OP and safety.

iirc I got the actual rate limits from having them too high and faulting the EPAS

bongbui321, 2024-07-18:
@robbederks, for context, I got panda blocked message from fuzzing the tx msg, not from a real scenario. This realistically shouldn't happen in the field since you mostly engage openpilot when the delta btw `apply_angle` and current `car steering angle` is not too  large (i.e when on a high road)

However, let say you first try to engage Openpilot when the current steering angle of the car is already larger than the rate limit, then the final `apply_angle` produced by OP would be clip to the current steering angle, which is > then the rate limit.  When it gets to panda, the last `desired_angle` by panda is 0 and the new `desired_angle` would be > the rate_limit -> causing a violation in panda.

sshane, 2024-08-20:
Going to close since we removed the Tesla port, heads up to @lukasloetkolben about this in his PR. I propose we just perform this limit before the rate limit

closed 2024-08-20 by sshane