commaai/openpilot #33007: Make `replay` tool run in realtime on a comma 3/3X

adeebshihadeh, 2024-07-17:
This should mostly be getting the frame decoding done in hardware.

`./replay --demo --dcam --ecam` will be used for validation. Realtime streaming obviously excludes the time taken to download the log and camera files.

jakethesnake420, 2025-06-13:
Got the decoder working but I am only able to convert the encoded hevc to nv12 ubwc format. Anyone know how to use the snapdragon display engine rotator to convert the frames?

jakethesnake420, 2025-06-14:
Just based on debugfs and the hardware is filling buffers with data:

This is a snapshot while the decdoer is opened in my script
```
cat /sys/kernel/debug/msm_vidc/core0/inst_ffffffda300e0000/info 
==============================
INSTANCE: ffffffda300e0000 (Decoder)
==============================
core: ffffffdaaca98000
height: 1216
width: 1952
fps: 20
state: 9
secure: 0
-----------Formats-------------
capability: Output
name : HEVC
planes : 1
type: Output
buffer mode : static
count: 4
size for plane 0: 7077888

capability: Capture
name : UBWC YCbCr Semiplanar 4:2:0
planes : 2
type: Capture
buffer mode : dynamic
count: 14
size for plane 0: 4972544
size for plane 1: 16384
-------------------------------
completions[9]: done
completions[10]: done
completions[11]: done
completions[12]: done
completions[13]: done
completions[14]: done
completions[15]: done
completions[16]: done
completions[17]: done
completions[18]: done
completions[19]: done
completions[20]: done
completions[21]: done
completions[22]: done
completions[23]: done
completions[24]: done
completions[25]: done
ETB Count: 1
EBD Count: 1
FTB Count: 28
FBD Count: 15
Pending buffer references
        buffer: 0x70c00000 fd[0] = 7 size 4972544 refcount = 3
        buffer: 0x72000000 fd[0] = 8 size 4972544 refcount = 2
        buffer: 0x72600000 fd[0] = 9 size 4972544 refcount = 2
        buffer: 0x72c00000 fd[0] = 10 size 4972544 refcount = 2
        buffer: 0x73200000 fd[0] = 11 size 4972544 refcount = 2
        buffer: 0x73800000 fd[0] = 12 size 4972544 refcount = 2
        buffer: 0x73e00000 fd[0] = 13 size 4972544 refcount = 2
        buffer: 0x74400000 fd[0] = 14 size 4972544 refcount = 2
        buffer: 0x74a00000 fd[0] = 15 size 4972544 refcount = 2
        buffer: 0x75000000 fd[0] = 16 size 4972544 refcount = 2
        buffer: 0x75600000 fd[0] = 17 size 4972544 refcount = 2
        buffer: 0x75c00000 fd[0] = 18 size 4972544 refcount = 2
        buffer: 0x76200000 fd[0] = 23 size 4972544 refcount = 2
        buffer: 0x78800000 fd[0] = 26 size 4972544 refcount = 2
```

jakethesnake420, 2025-06-18:
https://github.com/commaai/openpilot/pull/35569
https://github.com/commaai/openpilot/pull/35575

adeebshihadeh, 2025-06-18:
Nice! Don't worry too much about the organization/structure; I can help with that once you have a complete PR.

jakethesnake420, 2025-07-01:
I do have it working now but unfortunately at this point, it's a huge PR.  In it's current state, without trying to condense the code, it's about 1500 lines. I could probably get it down to 600 lines.

greatgitsby, 2025-09-06:
solved by https://github.com/commaai/openpilot/pull/35821

@jakethesnake420 claim it!

closed 2025-09-06 by adeebshihadeh