commaai/openpilot #33016: comma 3X: no panda due to invalid CR3 config

adeebshihadeh, 2024-07-19:
Notes about the issue: https://github.com/commaai/panda/commit/5ee262f3cbcc0bc780e015c3d9d101ef290cda3a.

## Recovery procedure

Prep:
* 2 USB-C cables
* install `master-ci` on your comma 3X: `installer.comma.ai/commaai/master-ci`
* setup your PC
  * `git clone https://github.com/commaai/agnos-builder`
  * download special firehose
* understand that OBD-C is port #1 and aux is port #2

[image]

Steps:
1. power device through OBD-C port for 1 minute
2. remove OBD-C power and plug into the aux port
3. run `tools/edl --loader=prog_firehose_`
4. wait 10 minutes, leaving everything connected
5. plug power into the OBD-C port
6. run `tools/edl reset && fastboot continue`
7. voila! all fixed

You might need to repeat the process a few times before it works.

closed 2024-07-19 by adeebshihadeh

connoralydon, 2024-07-19:
on Apple Silicon Mac. I had to use Rosetta 2 and prefix the toold/edl command with arch -x86_64 as well as pip install libusb.

connoralydon, 2024-07-19:
this worked, if you need some help or troubleshooting I'm @lydonfyi on X

itsTreble, 2024-07-19:
confirm this fixed no "no panda" issue caused by me flashing sunnypilot dev-c3 on my new c3x right out of the box. 

I had no luck doing this on a windows machine, even tried WSL. Alternative would be to create a linux VM or just make a bootable linux USB and run the recovery from there. I chose the latter and got my c3x back to "vehicle online"!

bravochar, 2024-07-23:
For anyone else trying to do this with WSL, I appear to be having some luck with [usbipd.exe](https://learn.microsoft.com/en-us/windows/wsl/connect-usb). I had an error when initially connecting to the AUX port to Windows, but I unplugged and flipped the USB-C connector (I know that shouldn't make a difference, but sometimes it does if cables/connectors are bad). After running `usbipd.exe bind --busid <bus-id>` in an Adminstrator Powershell and `usbipd.exe attach --wsl --busid 10-4` in a regular Powershell, I was able to see the device in my Ubuntu WSL shell with `lsusb`.

Provided that my device is recovered with this procedure, will I be able to install a custom branch again without invalidating the CR3 config? It's not clear from anything above what _caused_ the problem, and I'd hate to go through the trouble of fixing it only to cause it again. Should I rebase my custom branch off of `master-ci` instead, or are there any other steps I should take?

itsTreble, 2024-07-23:
> Provided that my device is recovered with this procedure, will I be able to install a custom branch again without invalidating the CR3 config? It's not clear from anything above what _caused_ the [...]

I was able to get panda back with the recovery steps, reset it then load dev-c3 sunnypilot without any issue.

adeeb has a summary what happened here:  https://github.com/commaai/panda/commit/5ee262f3cbcc0bc780e015c3d9d101ef290cda3a

AskAlice, 2024-08-02:
This is happening on my comma 3, not just 3x ig

makifb, 2025-02-04:
Can't pass the error 
Could someone please help
[image]