commaai/openpilot #33033: New support car docs

adeebshihadeh, 2024-07-20:
This is v2 of [CARS.md](docs/CARS.md). Currently, CARS.md shows all the cars supported by upstream openpilot. The goal of this new page is to be a one stop shop to see the status of any car's support.

 It'll include:
* every single car: cars supported in upstream, dashcam cars in upstream, cars supported in forks (even if they're forever in forks), cars we know won't work (no ACC/LKAS), cars with blockers (SecOc), etc.
* super granular columns: has fuzzy fingerprinting, steering torque, etc.
* some info will be populated by openpilot, and the rest will be done manually

This will be super information dense, so I imagine we'll need to move this to a webpage instead of a GitHub Markdown file.

These docs should also help make it clear when a car port is ready to supported and merged into upstream openpilot.

andiradulescu, 2024-07-22:
Does this invalidate https://github.com/commaai/openpilot/issues/32690?

adeebshihadeh, 2024-07-22:
No