commaai/openpilot #33041: C3 Screen Too Bright @ Night

jsegill, 2024-07-22:
### Describe the bug

Since installing 0.9.7 screen brightness at night on the C3 has been way too bright. Usually the night compensation for brightness works well, seems to have changed w/ 0.9.7. 

Whole route has the issue happening. 

### Provide a route where the issue occurs

4ecfcac756318cbf/00000035--99d3637ece

### openpilot version

0.9.7, release3

### Additional info

Fixing the brightness compensation is good but I would suggest adding an alternative brightness toggle. The user would be able to easily override screen brightness settings if they do not like stock brightness settings.

CruiseBrantley, 2024-07-23:
It's not an intermittent issue at all, for me at least. Pick any route at night with this dongle `149d6f706527cfe1`
Here's 1 example `149d6f706527cfe1/00000115--907e342143`
Essentially just stuck at 100% brightness it seems, I only notice it at night (*every* time) but possibly any route just not being noticeable during the day.

jkbkster, 2024-07-29:
Night driving only possible with manual brightness setting currently

wired4sound, 2024-07-29:
Commenting to say I am having this issue as well recently.

Blinding at night.   Manually turn down.  Then during day, can't see screen unless I shade it from the sun to turn it back up again because I cranked it down the night before.

adeebshihadeh, 2024-08-20:
Can you guys test `installer.comma.ai/commaai/fix-too-bright`?

MarcoTheDingo, 2024-08-21:
@adeebshihadeh looks like that helped! Not as dim as I personally keep my infotainment, but it was enough that it wasn't blinding. Route: 5130484aa8069bad/000000c9--4b7e4e53c6

MarcoTheDingo, 2024-08-27:
@adeebshihadeh are you looking for any other users to test, or any other routes to merge this?

adeebshihadeh, 2024-08-27:
So that isn't a mergable branch, it's just a test. Now that it's been tested, I'll come up with a mergable branch for you guys to test soon.

jsegill, 2024-08-27:
I can also test in a couple of days. Just really busy the last week...

jsegill, 2024-09-12:
Worked well on this route today `4ecfcac756318cbf/00000000--abb1191d75/0`

MarcoTheDingo, 2024-09-12:
The more I use this branch the more it seems like it's just darkening the input from the cameras and not the actual screen. If I open the sidebar or go into settings it's just as bright as not being on this branch.

Looking back at old footage from previous releases, the screen was a lot darker. Is there a way to reinvestigate this as an auto brightness issue instead of the gamma input on the cameras?

adeebshihadeh, 2024-09-12:
> Looking back at old footage from previous releases

What footage? You can't tell this from the device's logs, unless you mean recording a video of the device with your phone?

MarcoTheDingo, 2024-09-12:
Yes looking at old footage I recorded of the comma 3 and from my own memory, it was a lot dimmer.

jsegill, 2024-09-14:
Can confirm the UI is bright. Screen is still bright at night.

wired4sound, 2024-09-14:
Also confirming.  How does this go on so long? Lol, does it not get dark in San Diego?  It's unusable at night without cranking it way down.   Then you can't see it in the morning without covering it with your hand to make it dark enough to see the settings to turn it back up.

jsegill, 2024-09-18:
The screen is the brightest at midnight. This was at 2400-0100 w. `Screen-Too-Bright`. `4ecfcac756318cbf/00000016--7005cf51a2/0`

MarcoTheDingo, 2024-10-07:
Here's an example of auto brightness mostly working. Seems to work best when there's changes to the brightness. In this route we start in my garage, then go to a sunset drive into twilight. It starts to dim with the sunset correctly until I get on the highway in which it full brights for a mile or two, then starts to auto dim again correctly: 5130484aa8069bad/0000012a--3c9d04250a

MarcoTheDingo, 2024-11-13:
@adeebshihadeh confirmed this bug is fixed by #33984! Finally, my eyes are spared. 🎉

closed 2024-11-13 by adeebshihadeh