commaai/openpilot #33083: Minimize MetaDrive install size

adeebshihadeh, 2024-07-26:
MetaDrive is going to be a core part of the openpilot on PC experience soon, so most openpilot installations will want to have it installed. Since openpilot itself is small, this means MetaDrive

MetaDrive and its two of its dependencies, panda3D and cv2, are our top 3 largest dependencies combining for 691M.
```
(.venv) macbookair:site-packages adeebshihadeh$ du -hs * | sort -h | tail -n 5
 95M    pyarrow
130M    casadi
154M    cv2
241M    metadrive
296M    panda3d
```

In order to claim the bounty:
- [ ] PR to MetaDrive to move **all** dependencies we don't use into a separate dependency group
- [ ] can we make `panda3D` any smaller? if so, let's do the same thing that we're doing for MetaDrive
- [ ] do we need cv2? 
- [ ] once all minimizing is done, a good breakdown of where the remaining size is being spent (and whether its relevant to our application)

MetaDrive PRs should be made against [upstream MetaDrive](https://github.com/metadriverse/metadrive), however if they don't respond quickly, we can merge them into our fork.

maxime-desroches, 2024-07-26:
Most of panda3d are shared libraries. I did a quick test and for our current use case (the bridge test) we don't need:

- libp3tinydisplay.so
- libp3assimp.so
- libp3ffmpeg.so

those 3 alone are 33% of panda3d size

maxime-desroches, 2024-07-26:
So I was able to reduce the size of metadrive to 53M and ran it here https://github.com/commaai/openpilot/pull/33085/ 

Steps:
- remove unused render_pipeline -> 69M (might be hard to upstream)
- remove unused 3D models
- remove unused waymo + nuscenes datasets
- remove unused logos + splash screen
- quickly compress the textures: most textures are just raw 4K(!!!) or 2K images. There easily still 20M to gain be compressing other textures

adeebshihadeh, 2024-07-26:
From `uv pip tree`:
```
metadrive-simulator v0.4.2.3                                                                                        
├── requests v2.32.3 (*)                                                                                            
├── gymnasium v0.29.1                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── cloudpickle v3.0.0                                                                                          
│   ├── typing-extensions v4.12.2                                                                                   
│   └── farama-notifications v0.0.4                                                                                 
├── numpy v1.26.4                                                                                                   
├── matplotlib v3.9.1 (*)                                                                                           
├── pandas v2.2.2                                                                                                   
│   ├── numpy v1.26.4                                                                                               
│   ├── python-dateutil v2.9.0.post0 (*)                                                                            
│   ├── pytz v2024.1                                                                                                
│   └── tzdata v2024.1                                                                                              
├── pygame v2.6.0                                                                                                   
├── tqdm v4.66.4                                                                                                    
├── yapf v0.40.2                                                                                                    
│   ├── importlib-metadata v8.2.0                                                                                   
│   │   └── zipp v3.19.2                                                                                            
│   ├── platformdirs v4.2.2                                                                                         
│   └── tomli v2.0.1                                                                                                
├── seaborn v0.13.2                                                                                                 
│   ├── numpy v1.26.4                                                                                               
│   ├── pandas v2.2.2 (*)                                                                                           
│   └── matplotlib v3.9.1 (*)                                                                                       
├── progressbar v2.5                                                                                                
├── panda3d v1.10.14                                                                                                
├── panda3d-gltf v0.13                                                                                              
│   ├── panda3d v1.10.14                                                                                            
│   └── panda3d-simplepbr v0.12.0                                                                                   
│       ├── panda3d v1.10.14                                                                                        
│       └── typing-extensions v4.12.2                                                                               
├── pillow v10.4.0                                                                                                  
├── pytest v8.3.2                                                                                                   
│   ├── iniconfig v2.0.0                                                                                            
│   ├── packaging v24.1                                                                                             
│   └── pluggy v1.5.0                                                                                               
├── opencv-python-headless v4.10.0.84                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   ├── numpy v1.26.4                                                                                               
│   └── numpy v1.26.4                                                                                               
├── lxml v5.2.2                                                                                                     
[... cut ...]
│   ├── pyogrio v0.9.0                                                                                              
│   │   ├── certifi v2024.7.4                                                                                       
│   │   ├── numpy v1.26.4
│   │   └── packaging v24.1
│   ├── packaging v24.1
│   ├── pandas v2.2.2 (*)
│   ├── pyproj v3.6.1
│   │   └── certifi v2024.7.4
│   └── shapely v2.0.5
│       └── numpy v1.26.4
├── shapely v2.0.5 (*)
├── filelock v3.15.4
└── pygments v2.18.0

```

closed 2024-09-04 by maxime-desroches