commaai/openpilot #33092: Remove `control` dependency

adeebshihadeh, 2024-07-26:
It's just used in one place in one test: https://github.com/commaai/openpilot/blob/a1dce6ef475c69e446ab1ce13f1aadb26092804c/selfdrive/controls/lib/tests/test_vehicle_model.py#L50

The goal is to replace it with pure Python (and some numpy). 

`uv pip tree` for control:
```
control v0.10.0
├── numpy v1.26.4
├── scipy v1.14.0
│   └── numpy v1.26.4
└── matplotlib v3.9.1
    ├── contourpy v1.2.1
    │   └── numpy v1.26.4
    ├── cycler v0.12.1
    ├── fonttools v4.53.1
    ├── kiwisolver v1.4.5
    ├── numpy v1.26.4
    ├── packaging v24.1
    ├── pillow v10.4.0
    ├── pyparsing v3.1.2
    └── python-dateutil v2.9.0.post0
        └── six v1.16.0
```

Jaishree2310, 2024-07-27:
Hey @adeebshihadeh, Can you please assign this issue to me?
Thank you!

coderight1, 2024-07-29:
Hi @adeebshihadeh ,
I am very interested in working on this issue, and I have thoroughly reviewed the codebase. I am confident in my ability to resolve this issue. Here’s my approach:

1. **State-Space Replacement**: I will implement a custom `StateSpace `class using pure Python and NumPy to replace the dependency on the `control `library. This class will handle continuous and discrete-time systems, ensuring compatibility with the existing codebase.
2. **Integration and Testing**: I will integrate the new `StateSpace `class into `test_vehicle_model.py`, replacing the current usage of the control library. I will ensure that all existing tests pass and maintain the functionality of the original implementation.

Given my experience and this detailed plan, I am confident in my ability to address this issue effectively.
### Could you please assign this issue to me?

### Thank you!

closed 2024-07-29 by maxime-desroches