commaai/openpilot #33117: docs: setup glossary

adeebshihadeh, 2024-07-28:
This bounty is for both setting up the glossary infrastructure and populating it with most openpilot terms.

Requirements:
* a master definition file
* "glossary page" under the concepts section
* nice to have: click or hover on words defined in the glossary to get the definition on any docs.comma.ai page

wiki has a good starting point on definitions: https://github.com/commaai/openpilot/wiki/General-Terms

mawutory, 2024-07-30:
hi @adeebshihadeh want to take this up. please can you provide more details?

devSuryansh, 2024-08-15:
Hey @adeebshihadeh,
I'd like to work on this issue. Could you please assign this issue to me?

mawutory, 2024-08-15:
@devSuryansh you're free to work on it.

Pious1020, 2024-09-14:
hey, is the bounty for this still open?

dupelx, 2024-10-05:
hey can i work on this issue?

dupelx, 2024-10-06:
@adeebshihadeh ?

LouisVelez, 2024-12-06:
@adeebshihadeh whenever you have a chance please let me know if this meets your requirements #34155

PR #34231 (2024-12-14): docs: glossary infra
This PR replaces #34155 and splits the work needed for the glossary infra. This PR intention is to only set-up the infrastructure.

Remaining work:
- [x] Add term file #34232 
- [x]  Modify`glossary.md`#34234

Once This PR and the term file gets merged the modified `glossary.md` should be good to go. Will leave it in draft until the other 2 Prs are merged

Closes #33117
closed 2024-12-14 by adeebshihadeh