commaai/openpilot #33164: tools/replay/can_replay.py not working with Panda Jungle

ep1cman, 2024-08-01:
### Describe the bug

I am attempting to use the CAN replay script provided in `tools/replay/can_replay.py`. I have a panda jungle board connected to my PC and a Comma 3X connected to the panda jungle.

The ignition signal is correctly sent and detected by the Comma 3X but after this I face multiple issues:

Firstly I will often see this error which then hangs, forcing me to restart the script:
```
~/git/openpilot$ tools/replay/can_replay.py
Loading log...
carFingerprint (for hardcoding fingerprint): 'TOYOTA COROLLA TSS2 2019'
100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:04<00:00,  2.11s/it]
Finished loading...
starting send thread for 140020000d51333038363231
failed to get serial number of panda
Traceback (most recent call last):
  File "/home/seb/git/openpilot/panda/python/__init__.py", line 362, in usb_connect
    this_serial = device.getSerialNumber()
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 2019, in getSerialNumber
    return self.open().getSerialNumber()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 1132, in getSerialNumber
    return self.getASCIIStringDescriptor(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 1288, in getASCIIStringDescriptor
    received = mayRaiseUSBError(libusb1.libusb_get_string_descriptor_ascii(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorPipe: LIBUSB_ERROR_PIPE [-9]
```

When it does work, using the default route, the Comma 3X just displays the following message:
"openpilot Unavailable. Waiting for controls to start", however I can see the LEDs on the panda jungle flashing.

I also tried to use the demo route from tools/replay/replay (`a2a0ccea32023010|2023-07-27--13-01-19`), this gets me a bit further but now instead I see the message "Dashcam Mode. Car Unrecognized"

### OS Version

Ubuntu 24.04

adeebshihadeh, 2024-08-01:
Have you tried reflashing your jungle with the latest fw?

https://github.com/commaai/panda/tree/master/board/jungle

ep1cman, 2024-08-02:
I tried this but get the same results:

```
seb@ava:~/git/openpilot$ source .venv/bin/activate
(.venv) seb@ava:~/git/openpilot$ cd panda/board/jungle/
(.venv) seb@ava:~/git/openpilot/panda/board/jungle$ ./flash.py
scons: Entering directory `/home/seb/git/openpilot/panda'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
arm-none-eabi-gcc -o board/jungle/bootstub-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld -Iboard/jungle -Iboard -I. -Iboard board/bootstub.c
arm-none-eabi-gcc -o board/jungle/bootstub-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld -Iboard/jungle -Iboard -I. -Iboard board/bootstub.c
arm-none-eabi-gcc -o board/jungle/main-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld -Iboard/jungle -Iboard -I. -Iboard board/jungle/main.c
arm-none-eabi-gcc -o board/jungle/main-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld -Iboard/jungle -Iboard -I. -Iboard board/jungle/main.c
arm-none-eabi-gcc -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld -o board/jungle/obj/startup_panda_jungle.o -c board/stm32f4/startup_stm32f413xx.s
arm-none-eabi-gcc -o board/jungle/rsa-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld -Iboard/jungle -Iboard -I. -Iboard crypto/rsa.c
arm-none-eabi-gcc -o board/jungle/sha-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld -Iboard/jungle -Iboard -I. -Iboard crypto/sha.c
arm-none-eabi-gcc -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld -o board/jungle/obj/startup_panda_jungle_h7.o -c board/stm32h7/startup_stm32h7x5xx.s
arm-none-eabi-gcc -o board/jungle/rsa-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld -Iboard/jungle -Iboard -I. -Iboard crypto/rsa.c
arm-none-eabi-gcc -o board/jungle/sha-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld -Iboard/jungle -Iboard -I. -Iboard crypto/sha.c
arm-none-eabi-gcc -o board/jungle/obj/bootstub.panda_jungle.elf -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -Iboard/stm32f4/inc -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32f4/stm32f4_flash.ld board/jungle/obj/startup_panda_jungle.o board/jungle/rsa-panda_jungle.o board/jungle/sha-panda_jungle.o board/jungle/bootstub-panda_jungle.o
arm-none-eabi-objcopy -O binary board/jungle/obj/bootstub.panda_jungle.elf board/jungle/obj/bootstub.panda_jungle.bin
arm-none-eabi-gcc -o board/jungle/obj/bootstub.panda_jungle_h7.elf -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -Iboard/stm32h7/inc -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tboard/stm32h7/stm32h7x5_flash.ld board/jungle/obj/startup_ [... cut ...]
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/seb/git/openpilot/.venv/lib/python3.12/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorIO: LIBUSB_ERROR_IO [-1]
```

On a second attempt I don't get the above error but the Comma 3X says "openpilot Unavailable. Waiting for controls to start" - however I can see the LEDs blinking on the panda jungle for a short while

adeebshihadeh, 2024-11-08:
Still having issues?

closed 2024-11-08 by adeebshihadeh