commaai/openpilot #33176: "Process Not Running; uploader" on devices uploading rlogs

Quantizr, 2024-08-03:
### Describe the bug

With cafca4f89176855585835ff7c67dea5a9b08ca23, a "Process Not Running; uploader" alert appears when engaging after reboot around 1/3 of the time in the testing closet (which uploads rlogs as immediate_priority). Seems like it could be a race condition somewhere?

Issue does not occur with the previous commit 2cd9975e4fe31bf89cdbf41d3d5db02f519a2718.

[cafca4f89176855585835ff7c67dea5a9b08ca23_b9b916de_00000136--9ec02ae39d--0_rlog.zst.zip](https://github.com/user-attachments/files/16478778/cafca4f89176855585835ff7c67dea5a9b08ca23_b9b916de_00000136--9ec02ae39d--0_rlog.zst.zip)

### Provide a route where the issue occurs

attached

### openpilot version

cafca4f89176855585835ff7c67dea5a9b08ca23

deanlee, 2024-08-03:
It looks like the error happens when accessing a subdirectory of the log root, causing an exception. 
@adeebshihadeh, we need to revert https://github.com/commaai/openpilot/commit/cafca4f89176855585835ff7c67dea5a9b08ca23. However, I think we still need to address this issue to prevent cleanup failures due to inaccessible subdirectories

adeebshihadeh, 2024-08-03:
We have to catch this in CI.