commaai/openpilot #33207: Get MetaDrive simulator working on macOS

adeebshihadeh, 2024-08-06:
Must work as well as on Linux.

At a minimum, this means it must pass the CI test: https://github.com/commaai/openpilot/blob/master/tools/sim/tests/test_metadrive_bridge.py.

neehhaa06, 2024-08-09:
Hello @adeebshihadeh :)

> I think this one would be help to you

- Update Deprecated Libraries, Replace  'pkg_resources; if it’s deprecated. Use alternatives like 'importlib.metadata'

- Make sure the 'setup_create_bridge'  fixture properly sets 'test_duration' and integrates correctly with the test setup.

- Verify that 'MetaDriveBridge' is initialized with the correct parameters.

mfeinLearn, 2024-08-11:
Has anyone taken a stab at this yet?

Pious1020, 2024-09-14:
hey, can you assign me to this issue?

adeebshihadeh, 2025-02-08:
@maxime-desroches is this done?

jacobkg189, 2025-04-03:
Is this project still open?

DotLYHiyou, 2025-05-07:
Maybe an option is to rebuild it from scratch?

aaryouz, 2025-10-02:
I wanna work on this is this still open

vsd2027, 2026-02-08:
Good Day, 

I would like to work on this issue. 

Thanks and Regards, 
Vedvyas Danturi.

MaryamKurbanova, 2026-09-04:
Hello, I’m currently giving this issue a try.

jmy126, 2026-09-05:
Lock on #38729 has timed out (no progress since 2026-08-29).

Working PR: https://github.com/commaai/openpilot/pull/38764

Covers the known macOS MetaDrive blockers:
- `spawn` multiprocessing context + picklable bridge + panda3d GL 4.1 core
- skip `forkpty` / `unblock_stdout` on Darwin (manager otherwise dies silently)
- sim camera EOF via `time.monotonic()`
- ignore hardwared free-space gate under `SIMULATION`
- `test_spawn_safety.py` (local 4 passed; CI green including build macOS / unit tests / process replay 0 changes)

Happy to iterate on review feedback. Note `TestSimBridgeBase.test_driving` is still `@unittest.skip` on master, so the added spawn-safety coverage is the currently runnable verification path.

miniquinox, 2026-09-08:
I got MetaDrive running on Apple Silicon and drove the stack far enough to find out what actually blocks this. https://github.com/commaai/openpilot/pull/38815 fixes four bugs. One of them isn't macOS specific, and I think it's the reason nobody has been able to finish this issue or #30693.

**The sim's VisionIPC buffer is the wrong size, on every platform.** `tools/sim/lib/camerad.py` calls `create_buffers()`, which packs NV12 tightly at `W*H*3/2`. modeld reads the frame out of that buffer using camerad's padded layout via `nv12_copy_size`. For 1928x1208 that's 3,735,552 bytes against 3,493,536 allocated, so modeld dies on its first frame with `ValueError: buffer is smaller than requested size`. `process_replay.py:209-210` already allocates the padded layout, the sim doesn't. This fails identically on Linux.

The three macOS ones: `hardwared` reads `/proc/stat` and never starts, the sim shares multiprocessing objects across the default start method so a spawned child can't rebuild the semaphore, and camera frames are stamped with `frame_id * 0.05 * 1e9` instead of the `logMonoTime` clock, so locationd rejects every camera odometry pose.

With those fixed, on an M2 Max: all managed processes run, calibration reaches `calibrated` with 20 valid blocks, `sensorsOK` and `posenetOK` are true. It still doesn't engage, and the reason is a rate problem rather than a macOS one.

### The remaining wall, measured

The driving model is too slow for the stack on CPU, everywhere.

| configuration | per inference | rate |
|---|---|---|
| tinygrad `DEV=CPU`, JIT, M2 Max | 100.8 ms | 9.9 Hz |
| tinygrad `DEV=CPU`, uncompiled `OnnxRunner` | 348.9 ms | 2.9 Hz |
| under full sim load, `modelV2` as published | | 7.5 Hz |
| what the stack expects | 50 ms | 20 Hz |

The camera publishes at about 17 Hz and modeld consumes at 7.5, so `vipc_dropped_frames` is at least 1 on essentially every iteration. That makes `fill_model_msg.py:179` set `cameraOdometry.valid = False`, which makes locationd's `sm.all_valid()` false, which makes `deviceMotion.inputsOK` false, which raises `locationdTemporaryError` and blocks engagement forever. Nothing downstream recovers.

To confirm that chain rather than infer it, I paced the camera thread down to 7 Hz and re-ran. `inputsOK` went true, calibration held, and openpilot engaged and drove (`selfdriveState.active`). So the sim is otherwise healthy on macOS, and it's the model rate against the camera rate that stops it. The next thing to show up after that is `selfdrivedLagging`, since selfdrived can't hold its 100 Hz while MetaDrive renders 1928x1208 on the same box.

### Two hypotheses I tested and can rule out

It isn't the float16 dtype, at least not for tinygrad. The shipped `driving_supercombo.onnx` is a float16 graph: 315 of 334 initializers, 30,003,201 parameters, first node `Cast(img) -> FLOAT16`, 82 `Gemm` plus 8 `MatMul`. Upcasting the whole graph to float32 inside tinygrad's own ONNX reader, weights and `Cast` nodes and input specs, is worth 1.05x. It goes from 100.8 ms to 95.7 ms. Whatever fp32 buys an ONNX Runtime style GEMM kernel, tinygrad's CPU codegen doesn't have that cliff.

It isn't the render path either, on a real GPU. MetaDrive steps and renders at 33 fps here, and stripping the float HDR target and tonemap pass out of `RGBCamera._setup_effect` changes nothing, 30.2 ms against 30.3 ms. That win is specific to llvmpipe software rendering on a CI runner.

### The one that would fix macOS, and why I didn't ship it

`SConscript:35` builds Darwin with `DEV=CPU`. On Metal the same model is 2.7 ms, so 367 Hz. That's 38x faster and far inside the 20 Hz budget, and that single flag change would make the simulator work properly on every Mac.

It doesn't build though. `compile_modeld.py`'s pickle round trip check fails on Metal: the reloaded JIT computes something different from the original for the same seed, by max 86.75 against an output standard deviation of 8.77, so it isn't float noise. Plain tensors and a simple `TinyJit` do survive `dump_oob`/`load_oob` on Metal, so it looks specific to the model's JIT rather than to serialization in general. I left it alone rather than guess at tinygrad internals.

So the question is whether a GPU backend on Mac is the direction you'd want, or whether the simulator should instead pace its camera to whatever modeld can consume and accept running slower than real time on a slow machine. The second is a few lines and I have it working, but it changes what the sim means, so it feels like your call rather than mine. Happy to finish whichever one you point at.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

jmy126, 2026-09-09:
Updated https://github.com/commaai/openpilot/pull/38764 with the VisionIPC padded-buffer fix (modeld short-buffer on every platform), macOS SystemStats stub, and 7 Hz camera pacing on non-Linux so CPU modeld can engage. Building on the measurements in the thread above.

closed 2026-09-09 by adeebshihadeh