commaai/openpilot #33230: Get mypy passing on macOS

maxime-desroches, 2024-08-08:
currently fails with:

```bash
mypy..............................................[✗]
system/loggerd/xattr_cache.py:10: error: Module has no attribute "getxattr"  [attr-defined]
system/loggerd/xattr_cache.py:22: error: Returning Any from function declared to return "None"  [no-any-return]
system/loggerd/xattr_cache.py:22: error: Module has no attribute "setxattr"  [attr-defined]
common/realtime.py:32: error: Module has no attribute "sched_setscheduler"  [attr-defined]
common/realtime.py:37: error: Module has no attribute "sched_setaffinity"  [attr-defined]
system/athena/athenad.py:754: error: Module has no attribute "TCP_USER_TIMEOUT"  [attr-defined]
system/athena/athenad.py:755: error: Module has no attribute "TCP_KEEPIDLE"; maybe "TCP_KEEPALIVE" or "TCP_KEEPINTVL"?  [attr-defined]
Found 7 errors in 3 files (checked 517 source files)
```

sshane, 2024-08-08:
Looks like an out of date https://github.com/python/typeshed module

maxime-desroches, 2024-08-09:
Unfortunately no, `TCP_USER_TIMEOUT` and `TCP_KEEPIDLE` are Linux only constants and don't exists on MacOS

hellochang, 2025-02-08:
Will take a look. Could you assign the ticket to me? @adeebshihadeh

closed 2025-02-16 by maxime-desroches