commaai/openpilot #33301: Investigate raylib and others for ui

adeebshihadeh, 2024-08-14:
openpilot's UI used to be written in https://github.com/memononen/nanovg, and in retrospect that it was nicer to work in than . Qt is nice for cabana, but it turns out to not add much value to the openpilot ui and adds a bunch of conceptual and dependency complexity.

Some options
* https://www.raylib.com/
* https://github.com/Immediate-Mode-UI/Nuklear

We're looking for:
* cross platform: Linux, macOS, etc.
* small, simple. single header is ideal
* Python bindings?

The onroad camera view is a good test for trying out these libraries.

adeebshihadeh, 2024-08-14:
@deanlee want to try to make a small demo in raylib of the onroad camera view?

deanlee, 2024-08-16:
I tried writing a simple demo based on Raylib to display the road camera (without shaders). Raylib can simplify some OpenGL code, and its text rendering performance is also somewhat better (currently, in the drawHud function of the Qt UI, the three `drawText` function calls consumes about 3% of the CPU). 
Raylib might be a good option. The real challenge is the workload of migrating the entire UI to Raylib, including rewriting network management and WiFi manager. Additionally, Raylib itself lacks a signal-slot mechanism. All logic is driven by a while loop, and input/output would also need to be restructured, making the workload quite significant. Considering that the current UI is relatively stable, if the goal is just to save space, it might be worth first optimizing other dependencies that can be relatively easily removed before eventually migrating to Raylib, RayUI, or another library.

adeebshihadeh, 2024-08-16:
Qt was a big improvement over the old APKs, and I expect this to be a similar improvement. Smaller dependency, simpler library, better docs, etc. 

I think we do have to find some way to do this incrementally; like you said, it's too much to do all at once. If you can't figure out a way to do it, I can start it. Want to put up a draft PR with the demo you wrote?

deanlee, 2024-08-16:
I'll commit a draft after adding the shader, transformations, and other features to ensure it matches the CameraView's output.

AskAlice, 2024-08-18:
fyne is quite nice although it's written in golang which would be a bit of a hurdle, though it wouldn't be the first time the UI were separate from the py

deanlee, 2024-08-19:
PR #33326  provides a draft on using Raylib to display the road camera, with automatic scaling to fit the window while keeping the aspect ratio.

deanlee, 2024-08-25:
**First Step: Migrate Onroad Widgets to Raylib**
1.Organize the existing code by encapsulating related functions into specific classes to improve clarity and maintainability, making the transition to raylib smoother.
- [ ] Refactor CameraView to serve as a generic class for displaying vision streams, enhancing modularity and code reuse. https://github.com/commaai/openpilot/pull/33372
- [ ] Create a dedicated class for lead car updates and rendering functions.
- [ ] Create a dedicated class for driver monitor updates and rendering functions.
- [ ] Create a dedicated class for HUD updates and rendering functions.
- [ ] Create a dedicated class for model updates and rendering functions.
2. Migrate to raylib
- [ ] Port the rendering functions of the newly organized classes from Qt to raylib, ensuring better performance and seamless integration with raylib's rendering capabilities.

deanlee, 2024-12-12:
In my opinion, the current UI structure and display elements are relatively simple, making migration straightforward. The main challenges involve replacing the qdbus dependency with libdbus for the WiFi manager and implementing multi-language support. We need to create a tool similar to Qt’s lupdate to extract translatable strings and generate test cases. Once these tasks are completed, the migration will be quick and can likely be done in one go.

sunnyhaibin, 2025-01-11:
@mitchellgoffpc I see that you are working on porting UI to PyQt5 here: https://github.com/commaai/openpilot/tree/python-ui

Is this something you're experimenting with, or part of the roadmap?

mitchellgoffpc, 2025-01-11:
Just something I'm experimenting with

deanlee, 2025-01-12:
I’ve moved the discussion about the migration of wifimanager from [this comment](https://github.com/commaai/openpilot/issues/34271#issuecomment-2547520977) to this thread for easier discussion:

> Would switching to sd-bus with systemd be feasible? sd-bus provides a higher-level API and better integration with systemd, which could potentially simplify development. Alternatively, could we consider using something simpler like wpa_supplicant, or is libdbus still the necessary base for our implementation?

adeebshihadeh, 2025-01-12:
I’m open to it. Want to open a small example PR with it? I also found https://github.com/martinhaefner/simppl

deanlee, 2025-01-12:
Here is a demo WifiManager:  https://github.com/commaai/openpilot/pull/34364

deanlee, 2025-01-15:
Raylib doesn’t display special characters like ⌫, →, or non-English symbols correctly by default because it lacks a font manager like Qt. Qt’s QTextEngine automatically loads fonts for various languages and symbols, including control characters and emojis. 

In Raylib, fonts need to be manually loaded, and without the right ones, non-ASCII characters show up as "??". We need to implement a simple font management system to handle this, making the migration from Qt to Raylib smoother. Even without Raylib, removing Qt still means we must manage fonts manually. 

Anyone who can contribute would be greatly helpful.

adeebshihadeh, 2025-01-16:
@deanlee before we get too far in the Raylib rewrite, I think we need to check two more things:
- [ ] can we make a good scrolling demo (ideally without GPU)? Qt's performance is one of its biggest problems, and this isn't worth doing if it doesn't address this
- [ ] can we write the UI in python with raylib bindings? All python would be great, but even C++ widgets assembled with Python is fine too

adeebshihadeh, 2025-02-12:
@maxime-desroches showed me a good scrolling demo, so we're good to go now.

@deanlee I suggest we proceed like this:
1. Move over spinner, updater, and the rest of the auxiliary GUIs. These need to stay in C++ (and ideally statically linked) to avoid issues when upgrading AGNOS.
2. Start porting over the rest of `ui`. I think we want to do this in Python, but this means we won't be able to do it incrementally. But I think that's fine.

adeebshihadeh, 2025-02-12:
Actually maybe Python is fine for #1 too, and we just guarantee to always include a compatible Python raylib version in AGNOS.

deanlee, 2025-02-13:
> Actually maybe Python is fine for [#1](https://github.com/commaai/openpilot/issues/1) too, and we just guarantee to always include a compatible Python raylib version in AGNOS.

Using Python for all GUIs makes perfect sense. We can implement a Python-based WiFi manager that seamlessly integrates with both the main UI and auxiliary GUIs.

deanlee, 2025-02-14:
Move over spinner, updater, and the rest of the auxiliary GUIs:

- [ ] spinner & text window https://github.com/commaai/openpilot/pull/34583
- [ ] keyboard
- [ ] wifi manager
- [ ] updater

adeebshihadeh, 2025-02-16:
`installer` and `updater` do need to work with old versions of AGNOS, but we can save those for last.

adeebshihadeh, 2025-02-16:
Also now that we've merged a few of the UIs, I think it's time to restate the goals.

Goals: 
1. **Maintainability**: C++ -> Python, long build time -> no build time, big dependency -> minimal one for better portability. All huge wins.
2. **UI off the GPU**: https://github.com/commaai/openpilot/issues/30294
3. **Polish**: design aside, the current UI is janky with its poor scrolling, variable FPS, and a bunch of other little things. #1 will help a lot here too; more iterations means more polish.
---

@deanlee Given those goals, I think we should take the three UIs we've merged already and polish them up before continuing. This will set a good foundation for the rest of the rewrite:
* GuiApplication
  * make sure it's not using the GPU at all
  * add monitoring for the FPS and log when it's less than 60
  * add `DEBUG_FPS=1`: an environment variable for showing the FPS
  * add `STRICT_MODE=1`: kills the UI if it drops too much below 60 FPS. maybe master runs in this mode all the time?
* make the styles match the old UI
* make the scrolling in `text` feel like an iPhone. make an inertial scroller widget?

adeebshihadeh, 2025-03-16:
@deanlee are we ready to move the small UIs (spinner, text, etc.) to the new versions? It'd be a great step to start shipping some raylib UIs to production!

AskAlice, 2025-04-26:
gotta say during my job interview a few years ago with comma, my aversion to QT was a huge neg and now look at your position

closed 2025-09-27 by sshane