commaai/openpilot #33325: Openpilot C2 OnePlus 3T, can it run without installing a screen or canceling the screen? I have tried to enter the system without installing a screen

David-zzzzz, 2024-08-19:
### Describe the bug

Openpilot C2 OnePlus 3T, can it run without installing a screen or canceling the screen? I have tried to enter the system without installing a screen。
I plan to create a screenless version of OP, can I use code or trick the screen to make the system detect that the screen has been installed?
[image]

### Provide a route where the issue occurs

111111

### openpilot version

0.9

David-zzzzz, 2024-08-19:
I have tried to enter the system without installing a screen

adeebshihadeh, 2024-08-19:
This isn't a bug - check out our [Discord](https://discord.comma.ai) instead.