commaai/openpilot #33330: Plotjuggler --can not looking in dbc folder for dbc files

garrettpall, 2024-08-19:
### Describe the bug

When loading a route with plotjuggler, no dbc are shown on the lastest master unless they are copied into the higher level folder for opendbc.

### Provide a route where the issue occurs

N/A

### openpilot version

master

garrettpall, 2024-08-19:
[image]

sshane, 2024-08-20:
Is this WSL? Make sure the symlink is properly created for opendbc to opendbc_repo/opendbc

closed 2024-08-20 by sshane

garrettpall, 2024-08-20:
> Is this WSL? Make sure the symlink is properly created for opendbc to opendbc_repo/opendbc

Ubuntu 20.04 on a VM. If it’s just a me issue I’ll work around it

sshane, 2024-08-20:
Also try to update PJ with ./juggle.py --install, this fixed it for incognitojam