commaai/openpilot #33343: Intermittent CAN disconnections

Notstupe, 2024-08-20:
### Describe the bug

I randomly get CAN fault and comma power disconnect despite completely swapping hardware c3x included. Everything connected but car still fault and sometimes c3x give memory corruption error and forces me to reinstall completely.

f07256df14b3abd0/00000005--8f45df0486/0 none  good route
f07256df14b3abd0/00000006--057d476c7d/0

Lots of full logs updated to useradmin for check you guys though to see how intermittent it is. Could even be c3x bug.

### Which car does this affect?

Toyota prius 2021 LE

### Provide a route where the issue occurs

See above

### openpilot version

Release3

adeebshihadeh, 2024-08-20:
CAN2 is getting low level errors. You swapped all of the harness kit, OBD-C, and comma 3X?

Can you post a release3 route with the full route's rlogs uploaded?

Notstupe, 2024-08-20:
I purchase a new style harness/relay/comma which i ended up returning. This was purchased with my c3x. This my second vehicle so I have two c3x.  So swapped all comma hardware I've had laying around(been in since 03X days). So yes I've swapped everything at point. Comma power, ethernet cable, replay box, harness, usb C cable and c3x device. Still keep the error.

Notstupe, 2024-08-20:
f07256df14b3abd0/0000000f--00f1027056/0

Notstupe, 2024-08-20:
f07256df14b3abd0/00000009--79f535e946/0

Notstupe, 2024-08-20:
Everything should be uploaded in all the routes.

Notstupe, 2024-08-20:
f07256df14b3abd0/00000008--83cc931301/0

That should give you an routes with logs

Notstupe, 2024-08-21:
Additional note: I did have reset c3x panda on day of purchase via guide https://github.com/commaai/openpilot/issues/33016

This error doesn't show up when my car/c3x is extremely hot 90+f outside. But device is cool this error shows up.

Notstupe, 2024-09-23:
@adeebshihadeh @sshane  Should this issue stay here or should be opendbc. What can I do help you guys with this issue?

Notstupe, 2024-09-23:
Update routes on current version master-ci 9876830
f07256df14b3abd0/00000000--5c2b6e74f9 broken
f07256df14b3abd0/00000001--8b309dae30 working

adeebshihadeh, 2024-09-23:
Reach into support and they’ll get this taken care of.

Notstupe, 2024-09-23:
What type of issue is this then? C3x hardware issue?