commaai/openpilot #33418: Getting car unrecognized message on supported car

blkeku, 2024-08-31:
### Describe the bug

I am getting a “Car Unrecognized” error message. When I turn off the car I get this error message: openpilot was unable to identify your car. Your car is either unsupported or its ECUs are not recognized. Please submit a pull request to add the firmware versions to the proper vehicle. Need help? Join discord.comma.ai.

### Which car does this affect?

2018 Honda Civic

### Provide a route where the issue occurs

https://connect.comma.ai/27322c0540a97c51/00000001--1530a3dbe1

### openpilot version

0.9.7

### Additional info

Dongle ID: 27322c0540a97c51

adeebshihadeh, 2025-03-16:
Still an issue?

closed 2025-05-17 by adeebshihadeh

blkeku, 2025-05-18:
I am still having the issue. I had to switch to sunnypilot. But I would like to be able use openpilot

blkeku, 2025-05-18:
Please can you reopen the ticket. As my issue has not been resolved