commaai/openpilot #33636: CAN Error Check Connection on Ioniq 5 2022 with Hyundai Q harness

jimmy4345, 2024-09-24:
### Describe the bug

I have successfully installed the comma 3X with Hyundai Q harness on my Ioniq 5 2022 (no HDA 1 or 2 or 1.5) in Taiwan, Asia. After installed the stock openpilot (release3), it starts the calibration process.
However, in a sudden then said, “CAN ERROR check connections”.

I have checked all the cables several times. The device is new and undamaged.
I also test different branch like OP 0.9.6, 0.9.7 master-ci, every now and then I get the same error.

### Which car does this affect?

Ioniq 5 2022 (no HDA 1 or 2 or 1.5)

### Provide a route where the issue occurs

2024-09-20 08:53:09 | 57a2cea2bca4571d/00000004--8878429e70

### openpilot version

0.9.7

### Additional info

I have used some tools to concern if the CAN message is responding from car with Hyundai Q harness and yes it has the CAN-FD signal. (Using PCAN-USB)
I have tried to ssh in my device and when I implement the data/openpilot/panda/tests/can_printer.py, well start printing message and the status “CAN ERROR check connections” back to calibration process! But if I cancel the process then it came back to “CAN ERROR check connections”. (sometime work, sometime still maintain “CAN ERROR check connections”)

[image]

[image]

[image]

Also I have check the fw, although it shows Hyundai_Ioniq_5, but the fuzzy is true. ('fuzzyFingerprint': True)
I used "fingerprint_from_route.py" on openpilot and got the fingerprint 
"  **b'\xf1\x00NE1 MFC  AT EUR LHD 1.00 1.02 99211-GI010 211206'**  "
 which is not matching any other in Ecu.fwdCamera shows below (Ecu.fwdRadar are the same) 
[image]

There is only the LHD & RHD different.
Do I need to port it or any possible reason for “CAN ERROR check connections” or what I can do?
Is it possible the OBD-C cable problem?

jimmy4345, 2024-09-27:
This issue has been resolved through the discord thread "2022 Ioniq 5". Thanks.

jimmy4345, 2024-09-27:
Sorry for the mistake close. 

This issue has discussed at discord Hyundai-kia-genesis channel in "2022 Ioniq 5" thread.
It seems to flip the wires, and everything were fine. 
However, since I flipped the wires, that means openpilot is "not directly" supported.
Is this still need to be check?

Doro0, 2024-12-08:
Hi jimmy I had the same problem. I switched out the harness with k and it work flawless