commaai/openpilot #33667: longitudinalPlan: Not speeding up smoothly or all the way to set speed

garrettpall, 2024-09-28:
### Describe the bug

**Issue**
Car struggles to accelerate smoothly when overtaking a car. Car also does not speed up to set speed when not lead visible.

**Attempted Troubleshooting**
Reflash (flash.comma.ai) and reinstall
Turn off radar tracks and run VOACC
Update to latest master-ci
Run Longitudinal Maneuvers Testing Tool

**accel Pulsing**
[image]
[image]

**vEgo not hitting setSpeed**
[image]

**Disclaimer**
My car does not have long support upstream due to using extra panda, but code is still nearly identical to camera cars. Code changes should not have this affect on longitudinal plan (https://github.com/garrettpall/openpilot/commits/gm-long-test/)

### Provide a route where the issue occurs

162796f1469f2f1b/00000001--b37821a3c3

### openpilot version

3095fa79279e8c6424571a0195d04f1fcce579fa

garrettpall, 2024-09-29:
After a bunch of looking around, I have determined the culprit to be [limit_accel_in_turns](https://github.com/commaai/openpilot/blob/edf9522bc064aa30b348e53c674b052c860cc9fc/selfdrive/controls/lib/longitudinal_planner.py#L34). My steering angle is not 0 when driving straight but actually ~5.5 degrees (I probably need an alignment). limit_accel_in_turns looks only at sm['carState'].steeringAngleDeg and not angleOffsetAverageDeg. Peaks in accel are resulting from when limit_accel_in_turns is not limiting and accel drops to zero when it is limiting.

**Accel and Steer Angle**
[image]

**Measured and actual steer angle**
[image]

garrettpall, 2024-09-29:
At 80 mph, this is the limits I calculated for my car:
```Steering Angle: 5.0, Accel allowed: 1.18
Steering Angle: 5.1, Accel allowed: 1.08
Steering Angle: 5.2, Accel allowed: 0.97
Steering Angle: 5.3, Accel allowed: 0.84
Steering Angle: 5.4, Accel allowed: 0.69
Steering Angle: 5.5, Accel allowed: 0.49
Steering Angle: 5.6, Accel allowed: 0.09
Steering Angle: 5.7, Accel allowed: 0.00
Steering Angle: 5.8, Accel allowed: 0.00
Steering Angle: 5.9, Accel allowed: 0.00
Steering Angle: 6.0, Accel allowed: 0.00
Steering Angle: 6.1, Accel allowed: 0.00
Steering Angle: 6.2, Accel allowed: 0.00
Steering Angle: 6.3, Accel allowed: 0.00
Steering Angle: 6.4, Accel allowed: 0.00
Steering Angle: 6.5, Accel allowed: 0.00
Steering Angle: 6.6, Accel allowed: 0.00
Steering Angle: 6.7, Accel allowed: 0.00
Steering Angle: 6.8, Accel allowed: 0.00
Steering Angle: 6.9, Accel allowed: 0.00```

As seen above, we were definitely going above and below the 5.5 is degree cutoff

jyoung8607, 2024-10-01:
@garrettpall Thank you for the report! This is exactly the sort of thing we hope to flush out with the longitudinal maneuver test tool.

@haraschax This looks like a real issue to me. The angle offset isn't necessarily horrible, depending on platform. We don't even complain until the [offset reaches ten degrees](https://github.com/commaai/openpilot/blob/216145d4bc035e2bd7f10d9f3968631db8535578/selfdrive/locationd/paramsd.py#L24). I tried to report on the distribution but the commaCarSegments dataset doesn't contain  liveParameters.

In addition to the uncalibrated usage in [limit_accel_in_turns](https://github.com/commaai/openpilot/blob/d826b08369555f7e39d03511efa4ddea62701517/selfdrive/controls/lib/longitudinal_planner.py#L134), there's another uncalibrated usage in [paramsd](https://github.com/commaai/openpilot/blob/d826b08369555f7e39d03511efa4ddea62701517/selfdrive/locationd/paramsd.py#L109) which probably isn't doing us any favors. I'm not filing a PR right away because this will require some careful validation even if correct.

**This is a very easy bug to write.** It might be a good idea to sweep over xx and make sure nothing like this has crept into training or testing, something that might manifest later as ping-pong or hugging.

jyoung8607, 2024-10-01:
> but there's another uncalibrated usage in paramsd which probably isn't doing us any favors

Correction, I think this usage is probably intentional, but it wasn't immediately obvious and I still think it's an easy bug to write.

garrettpall, 2024-10-01:
For what it’s worth, this is what I am running now to fix it: https://github.com/garrettpall/openpilot/commit/894f2afca56b4f908abfba957da150c7692b2150

@twilsonco wrote this commit

PR #33827 (2024-10-22): Remove steering wheel offset for planner slow down for curves
**Description**

In ``selfdrive/controls/lib/longitudinal_planner.py``, the 	`limit_accel_in_turns()` function limits long accel based on lat accel. It uses a back of the hand calculation of lateral accel based on steering angle that does not include a car's steer angle offset (as observed by `liveParams`). In cars with a large steer offset, the `carState` steer angle can be high enough *when traveling in a straight line* that this function ends up limiting long accel. This problem is reported in #33667.

Additionally,  `limit_accel [... cut]
closed 2024-10-22 by sshane