commaai/openpilot #33721: simulator: sensors aren't filled

sshane, 2024-10-03:
The IMUState is never updated, so locationd is getting 0s for the sensor data, and then paramsd learns a bad offset, preventing from merging a fix to the steering angle: https://github.com/commaai/openpilot/pull/33718

https://github.com/commaai/openpilot/blob/72e19ccfc6de0a145bf778fdc701878f969347bb/tools/sim/lib/common.py#L30-L34

schmxtz, 2024-10-15:
I'm wondering what to do with the gyroscope value for IMUState. Metadrive does neither offer a simulated value for gyroscope nor for direction. We can just get the bearing value in [this](https://github.com/commaai/openpilot/blob/af727d41bbc094da7c56aa38e0526019125b935f/tools/sim/bridge/metadrive/metadrive_world.py#L83) function and store it in the state.imu.bearing variable. I guess the acceleration can also be calculated based on the last timestamp and last stored velocity and current velocity. But the gyroscope has to be kinda hacked, to be calculated based on the acceleration and direction the car is heading and even then it is really inaccurate. Or is it just that the value should stay zero?

sambhavnoobcoder, 2024-10-19:
hey @sshane , i would like to take up this issue . i think i have a code in line with @schmxtz 's suggestion that would solve this issue . if your okay with it , i'd like to work on this and resolve it asap .

zarisss, 2025-06-13:
Hey

Just following up on  PR (https://github.com/commaai/openpilot/commit/7cd2bdaf300d66d0462a8762acf57c652ce5c558) — it fixes IMU sensor readings(https://github.com/commaai/openpilot/issues/33721).
I've addressed the feedback and ensured everything passes the checks.

Let me know if there’s anything else you'd like me to tweak — happy to iterate!

Thanks again