commaai/openpilot #33758: GM: Broken Wheel Button Logic

nworb-cire, 2024-10-09:
### Describe the bug

Sometime since the opendbc migration, GM's steering wheel button logic has become broken. The resume button is now functionally equivalent to the set button. By this I mean that it will always cause the speed to be set to `max(65 if experimental else 25, vEgo)` instead of the previous set point as is desired.

### Which car does this affect?

Chevrolet Volt 2018

### Provide a route where the issue occurs

806907cd33ef2647/0000001b--0c38edf676/0

### openpilot version

596d8b13bbd82fdbc728d7b123c22f43e7b257f5

### Additional info

I have confirmed with a few other GM users that they experience this bug on master(-ci). You can see in the first segment of this route the broken behavior.

(Note: dirty git required to remove my car from dashcam mode)

jyoung8607, 2024-10-11:
Not GM specific, just had this on my Golf

adeebshihadeh, 2024-10-17:
@sshane seems to be this: https://github.com/commaai/openpilot/pull/33693. It now waits for a loopback on the enabled, and there's no more buttonEvents.

garrettpall, 2024-11-30:
Any progress on this?

garrettpall, 2025-01-08:
It’s been over a month with no comments/progress, thought I’d ping again.

morrislee, 2025-01-09:
Would love to have this fixed as well, along with the GM throttle issue, this makes long control unusable even if the model is improved

closed 2025-01-15 by sshane