commaai/openpilot #33800: Develoepr panel in settings

adeebshihadeh, 2024-10-15:
This should make it super easy to run the new longitudinal maneuver report and joystick mode.

It should have:
* SSH key setting (move from the Network tab)
* [Joystick mode](https://github.com/commaai/openpilot/tree/master/tools/joystick)
* [longitudinal maneuver report mode](https://github.com/commaai/openpilot/tree/master/tools/longitudinal_maneuvers)

Joystick and longitudinal maneuvers should be hidden on release branches.

jyoung8607, 2024-10-15:
* Start and stop ZMQ bridge

adeebshihadeh, 2024-10-15:
We should also think about moving the "openpilot long alpha" toggle. I'm leaning towards moving it.

AarjavPatni, 2024-10-19:
Should the joystick mode provide an option for the user to choose between comma and laptop? Or should it detect whether the comma is plugged in?

I'm not sure how that should look like. A description would be great.

AarjavPatni, 2024-10-20:
@adeebshihadeh Are we adding the option to start and stop ZMQ bridge or is it TBD? What about the "openpilot long alpha" toggle?

adeebshihadeh, 2024-10-28:
Ah yeah, let's move that one too. @BBBmau or @AlexandreSato want to move that?

closed 2024-10-28 by adeebshihadeh

AlexandreSato, 2024-10-29:
> Ah yeah, let's move that one too. @BBBmau or @AlexandreSato want to move that?

moved in #33885

AlexandreSato, 2024-11-11:
@adeebshihadeh 

#33885 Also fix two design issues in #33803 :
1 - the release toggles would only disappear after ignition on the release branch;
2 - cars without longitudinal control could activate longitudinal maneuvers.