commaai/openpilot #33813: Refactor tools/replay: Eliminate Qt Dependency

deanlee, 2024-10-18:
Refactors the tools/replay to eliminate its dependency on the Qt framework, making the tool lighter and more portable.

- [ ] Replace QCommandLineParser with getopt for command-line argument parsing.
- [ ] Replace HttpRequest with a libcurl implementation to load route info from https://api.commadotai.com/.
- [ ] Remove the use of Qt's signal/slot mechanism and replace it with standard C++ function callbacks.
- [ ] Replace QThread with std::thread to manage threading in a platform-independent way.
- [ ] Replace Qt-specific data types (QString, QVector, etc.) with standard C++ data types (std::string, std::vector, etc.).

closed 2024-10-19 by maxime-desroches

maxime-desroches, 2024-10-22:
Oh sorry I didn't realize that I closed this by accident!

deanlee, 2024-10-22:
:D I made the connection by accident.