commaai/openpilot #33814: No panda after switching branches

goodchoom, 2024-10-18:
### Describe the bug

After switching branches from mlsim_v4 to master_ci, openpilot was unable to go onroad due to the no panda error. After turning the car off and rebooting the device to no avail, I then unplugged the cable then plugged it back in after waiting 30 seconds, which seems to have resolved the issue.

### Provide a route where the issue occurs

ed882557605319f9/00000113--d26a86efa7

### openpilot version

074f1c98cb869ca65b9a511e830711cd6d6b73bd

BBBmau, 2024-10-18:
had the same experience when updating to the latest commit on master. No branch switching. Another reboot was needed before the `Panda Error` was resolved.

adeebshihadeh, 2024-10-18:
Seems to be related to the panda and `hardwared` setting up the amp. I'm thinking it's a race condition that the boot time speedup in AGNOS 11 is causing us to hit now. There's a libusb later (FD leak?), but that looks separate.

I reverted it for now, but I'll keep this open until it's merged again and root caused.

adeebshihadeh, 2024-10-24:
Easy to repro: run amp config in a loop and `Panda().reset(reconnect=True)`

closed 2024-10-31 by adeebshihadeh