commaai/openpilot #33840: panda siren on ignition off

sshane, 2024-10-21:


BBBmau, 2024-10-21:
This has been happening for quite a while for me also. I never made an issue due to not being sure of how to reproduce it.

adeebshihadeh, 2024-11-12:
What are your dongle IDs?

BBBmau, 2024-11-12:
9a31d18cd1a79c69

Sliicy, 2024-11-21:
I have this issue as well sometimes, albeit on SunnyPilot 0.9.7.1 (I know, not stock).

royjr, 2024-11-21:
I see this when restarting op/device while car still on.

adeebshihadeh, 2024-11-21:
> I see this when restarting op/device while car still on (and possibly engaged).

That's expected.

adeebshihadeh, 2025-01-22:
> I see this spammed when the car turned off, the can packet dropped out, and it reports 500+ can core resets, so perhaps the siren was valid here?

That's just because the ECUs on the bus go offline before we finish TX'ing all our msgs; it's fine that the TX fails. Those are some pretty low level logs. 

> [@adeebshihadeh](https://github.com/adeebshihadeh) Want me to look into this further?

Yes

sshane, 2025-01-28:
This is likely caused by https://github.com/commaai/panda/pull/1502. On the two routes in the original issue above, both display `transmitErrorCnt`s above 127 (actually both 135), fulfilling one part of the check in that PR. The other is that there is an ACK error, which could be likely given the car has been turned off. Then the CAN core reset is triggered >500 times over ~1 second.

When it recovers, lastDataError is 7 (noChange), but the transmit errors persist. Why does it take so long to recover I wonder? Should we also clear `transmit_error_cnt` when we clear the can core?

On a desk, when this condition is true on ignition off, pandaStates no longer is published (so panda doesn't get heart beat), and the siren sounds.

@briskspirit is this reset needed now that we clear the can core on multiplexing switch? (https://github.com/commaai/panda/pull/1967)

closed 2025-01-30 by sshane