commaai/openpilot #33850: Ford: resume from stop doesn't work with stock long

sshane, 2024-10-22:


BBBmau, 2024-10-27:
I'm experiencing this also except it's on experimental. This is from master on commit af73d6084d5ac77935282442c909432eebd676d1 i believe.

sshane, 2024-10-28:
Can you post a route where this happens?

BBBmau, 2024-10-28:
@sshane should be present here, if not I can get a more recent route

9a31d18cd1a79c69/00000001--79aa41f7e0/53

https://connect.comma.ai/9a31d18cd1a79c69/00000001--79aa41f7e0/3225/3237

sshane, 2024-10-28:
The model in experimental mode failing to resume is different than controlsd setting the carControl.cruiseControl.resume signal BTW, that's what this issue is for

BBBmau, 2024-10-28:
> The model in experimental mode failing to resume at traffic lights is different than controlsd setting the carControl.cruiseControl.resume signal based on the planner/lead BTW, that's what this issue is for

yeah the route is wrong, I'll be providing the route with the actual issue later today. Sorry for the confusion.

BBBmau, 2024-10-29:
@sshane looking to see what needs to be changed. A bit interesting how on resume it goes to the saved speed but is quickly reverted to be 65

[image]

sshane, 2025-01-15:
This works on a recent route, not sure if this was a real issue. Closing

closed 2025-01-15 by sshane