commaai/openpilot #33878: `panda fault: registerDivergent` on 3X

adeebshihadeh, 2024-10-26:
|    | fullname                              |   length |   maxqlog | platform             | git_commit   | version   |   devicetype | events                         |
|---:|:--------------------------------------|---------:|----------:|:---------------------|:-------------|:----------|-------------:|:-------------------------------|
|  0 | b3c7f01bb05086ee/000001c3--667a1483c0 |        0 |         5 | RAM_1500_5TH_GEN     | f11e549      | 0.9.8     |            8 | panda fault: registerDivergent |
|  1 | b3c7f01bb05086ee/000001c4--eee7ce41ee |        8 |        14 | RAM_1500_5TH_GEN     | f11e549      | 0.9.8     |            8 | panda fault: registerDivergent |
|  2 | 09ed4c7e7b4937fb/0000023b--bc2221b611 |        2 |         9 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  3 | 09ed4c7e7b4937fb/0000023c--907c27485e |        0 |         1 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  4 | 09ed4c7e7b4937fb/0000023d--5d754f4e25 |        6 |        19 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  5 | 09ed4c7e7b4937fb/0000023e--7604d7554b |        0 |         9 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  6 | 09ed4c7e7b4937fb/0000023f--9fbb38b3a3 |        3 |        14 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  7 | fea39f11afb2d0d0/000002ef--98fddaf0aa |        8 |        10 | HYUNDAI_SONATA       | 6175106      | 0.9.8     |            8 | panda fault: registerDivergent |
|  8 | 09ed4c7e7b4937fb/00000240--dee02f5e09 |        2 |         7 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
|  9 | 09ed4c7e7b4937fb/00000241--6d08d4167b |        0 |         1 | KIA_EV6              | 9ad2333      | 0.9.8     |            8 | panda fault: registerDivergent |
| 10 | 9534577804069efa/0000008a--fb1b1e46d2 |        7 |        13 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 11 | 9534577804069efa/0000008b--5ddf618b23 |        0 |         0 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 12 | 9534577804069efa/0000008c--e14a8e2bd3 |        0 |         1 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 13 | 6483a59cd49b6650/00000031--bc75b89976 |        0 |         1 | TOYOTA_CAMRY_TSS2    | 6175106      | 0.9.8     |            8 | panda fault: registerDivergent |
| 14 | 9534577804069efa/0000008d--8eee4597f8 |        2 |         7 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 15 | 9534577804069efa/0000008e--ee73a5c8de |        1 |        11 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 16 | 9534577804069efa/0000008f--9c462df786 |        6 |        14 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 17 | 9534577804069efa/00000090--a288f743c1 |        9 |        17 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 18 | 9534577804069efa/00000091--ff02133f1d |        7 |        15 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 19 | 2d429f425f438d36/0000016e--9d98cce6dc |       11 |        12 | TOYOTA_RAV4          | 6175106      | 0.9.8     |            8 | panda fault: registerDivergent |
| 20 | 9534577804069efa/00000092--4dd26aff6c |        7 |        15 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 21 | 9534577804069efa/00000093--d2d4d72421 |        0 |         0 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 22 | 9534577804069efa/00000094--300e1f8256 |        9 |        16 | CHEVROLET_BOLT_EUV   | 277691c      | 0.9.8     |            8 | panda fault: registerDivergent |
| 23 | f55d0ef8a18e4ffd/0000000c--622e0b53f8 |        4 |        12 | KIA_SPORTAGE_5TH_GEN | 30853a2      | 0.9.8     |            8 | panda fault: registerDivergent |
| 24 | f55d0ef8a18e4ffd/0000000d--8ffaff3e16 |       15 |        31 | KIA_SPORTAGE_5TH_GEN | 30853a2      | 0.9.8     |            8 | panda fault: registerDivergent |
| 25 | f55d0ef8a18e4ffd/0000000e--b83e953e1b |        2 |        10 | KIA_SPORTAGE_5TH_GEN | 30853a2      | 0.9.8     |            8 | panda fault: registerDivergent |
| 26 | f55d0ef8a18e4ffd/0000000f--3749f8b4cb |        4 |        16 | KIA_SPORTAGE_5TH_GEN | 30853a2      | 0.9.8     |            8 | panda fault: registerDivergent |
| 27 | cc0d52811074024e/00000089--a498859c22 |        0 |         1 | FORD_ESCAPE_MK4      | 9674c7b      | 0.9.8     |            8 | panda fault: registerDivergent |

sshane, 2025-05-27:
@robbederks is this fixed?

robbederks, 2025-05-27:
Looked at the kibana logs again earlier today, and haven't seen anything come in for 3X yet since we added it. Will keep monitoring it once in a while

closed 2025-05-27 by sshane