commaai/openpilot #33934: radard: calculate lead acceleration on cluster of tracks

sshane, 2024-11-05:
Right now we track lead accel per reported radar track. Lead accel is very important for the planner during high lead accel/decel situations.

We are fortunate that Toyota's reported radar tracks are mostly self consistent, in that a position/lead in space doesn't switch very often between the messages. However, it does happen. Here we could have reached 1 m/s^2 20% faster if the lead accel did not oscillate/start from 0:

https://connect.comma.ai/a2bddce0b6747e10/0000030f--995be3b11f/153/182

[image]

sshane, 2024-11-14:
Another case `a2bddce0b6747e10/000003eb--0f891e5a10/9`:

[image]

sshane, 2024-11-19:
Another case

https://connect.comma.ai/57048cfce01d9625/0000034a--9f0d6fc36b/103/130

[image]

sshane, 2024-12-15:
Another example on Toyota: https://connect.comma.ai/6ac3968bd1f66985/0000003b--e11775e688/430/482

[image]

sshane, 2025-01-27:
Another slow start from this: d9b97c1d3b8c39b2/0000002b--a0ecfaf597/71

adeebshihadeh, 2026-10-17:
Radar is going away soon!

closed 2026-10-17 by adeebshihadeh