commaai/openpilot #33976: Toyota: verify start from stop change

sshane, 2024-11-09:
https://github.com/commaai/openpilot/pull/33974. On most of the TSS2 platforms:

- [ ] TOYOTA_AVALON_TSS2
- [ ] TOYOTA_CAMRY_TSS2
- [ ] TOYOTA_SIENNA_4TH_GEN
- [ ] LEXUS_IS_TSS2
- [ ] LEXUS_RX_TSS2
- [ ] LEXUS_NX_TSS2
- [ ] TOYOTA_HIGHLANDER_TSS2
- [ ] TOYOTA_COROLLA_TSS2
- [ ] LEXUS_LC_TSS2
- [ ] TOYOTA_RAV4_TSS2_2023
- [ ] TOYOTA_RAV4_TSS2_2022
- [ ] LEXUS_ES_TSS2
- [ ] TOYOTA_RAV4_TSS2
- [ ] TOYOTA_PRIUS_TSS2
- [ ] TOYOTA_MIRAI
- [ ] TOYOTA_RAV4_PRIME
- [ ] TOYOTA_ALPHARD_TSS2
- [ ] TOYOTA_CHR_TSS2

devimauz, 2024-11-18:
Hi @sshane .
I still having problem with new long. Please check my drive. 4ffdd3a6a974b54b/00000003--9a2ea7cda2/11
Corolla TSS2 Hybrid EU.

sshane, 2024-11-18:
Thanks for the feedback, but this issue is only for the start from stop change we recently made.

sshane, 2025-01-14:
Haven't seen any bad cases/reports!

closed 2025-01-14 by sshane

devimauz, 2025-01-14:
> Haven't seen any bad cases/reports!

how many bad report should i do?

sshane, 2025-01-14:
This issue is about start from stop. We also merged some braking overshoot prevention last month, please provide a route on the latest version of master with issues and I can take a look