commaai/openpilot #33988: Unable to activate OP while brake hold active

jsegill, 2024-11-11:
### Describe the bug

When brake hold is enabled and in effect on a Toyota Corolla HB 2019, OP will not engage. 

Before 0.8.11, OP would engage while in a brake hold with a lead car. Not allowing OP to engage while at a stop, makes stop and go traffic disengagements more frustrating. One cannot simply re-enable OP after making a correction resulting in a stop of the car. Also trying to re-enable OP with brake hold active, results in the brake hold being released with no OP engagement. 

Stock Toyota system allows for ACC engagement while brake hold is active. Stock Toyota ACC requires a second push of the resume button to resume ACC functions.

I believe this stems from `PR commaai/openpilot#22810 ` to fix issue `Issue commaai/openpilot#22791 `. Anyway to prevent the integral build up, mentioned in `Issue commaai/openpilot#22791 ` so OP may be enabled again when brake hold is active? 

### What hardware does this issue affect?

comma three, comma two

### Which car does this affect?

Toyotas; Toyota Corolla HB 2019

### Provide a route where the issue occurs

99725b56a15b5510|2021-12-01--15-16-19	

### openpilot version

0.8.11

jsegill, 2024-11-13:
> Yes, just remove the alert that has a noEntry and show the `resumeRequired` alert instead. Transferred back since all changes need to be made in openpilot

Like this?   #`94e7c240c4dfa49aa70ab8a5ed4a59cf62316dd3` Commit 94e7c24

sshane, 2024-11-15:
Yes, exactly

jsegill, 2024-11-15:
I'll test Sunday and submit a PR if all goes well.

sshane, 2024-11-15:
You can open the PR now, we need to test on a few brands here to make sure it all works

jsegill, 2024-11-15:
Just got it in.

jsegill, 2024-12-31:
Last PR submitted won't work. Will need further work.

closed 2025-01-15 by sshane