commaai/openpilot #33993: deleter: inconsistency after AGNOS11 update

AlexandreSato, 2024-11-12:
### Describe the bug

I suspect that the deleter after AGNOS11 update is no more consistent. I haven't tested it yet to confirm, but I've already caught four people who had to delete realdata/* manually (because a noEnter "out of storage" event)

### Provide a route where the issue occurs

hdhd

### openpilot version

13760968bbdec44c2bf234fde79f8ef0e5bbe254

### Additional info

Not confirmed

adeebshihadeh, 2024-11-12:
Got any routes or dongles?

AlexandreSato, 2024-11-12:
> Got any routes or dongles?

only some dongs, added to description

nelsonjchen, 2024-11-13:
Got this error on this route:

https://connect.comma.ai/fe18f736cb0d7813/0000040d--3a202cd3b9

It actually increased by 1% after I couldn't engage anymore either. Logs/data are uploading. 

I pretty much only run master-ci so this was surprising.

AlexandreSato, 2024-11-13:
got one segment when it's happened: 
075b133b6181e058/000000cc--7d1fff3f36/42

Interestingly when I connected to the cell phone hotspot to ssh and run the deleter test script, the deleter worked the moment the connection was established without me running any command and the out of storage event cleared, when I get home I upload the route

korhojoa, 2024-11-17:
My routes don't appear in connect so can't really help there, but got it with dongle a954cbd0682c58bd

michaelhonan, 2024-11-18:
Yup mine's doing the same too. Hoping the new build of AGNOS in master will address...

My dongle ID is likely useless as I don't have a data connection anymore with my C3.
`dc30ef2ad294a732`

AlexandreSato, 2024-11-18:
I also have not seen that error too.

AlexandreSato, 2024-11-18:
grab more one: a0e5673b89222047/0000003d--2ac5e373e1/8

pbassut, 2024-11-23:
Not sure if this is related but my C3X wasn't uploading anything. After deleting realdata it started uploading again.

adeebshihadeh, 2024-11-25:
Just started looking into this:

`075b133b6181e058/000000cc--7d1fff3f36/42`
* eventually recovers towards the end of the drive, almost like the deleter was stuck
* no rlogs

AlexandreSato, 2024-11-25:
I missed it when I did the AGNOS flash in the sound issue test

AlexandreSato, 2024-11-29:
got some routes: 

075b133b6181e058/0000003e--67e218c522/1
075b133b6181e058/0000003d--2e512616b9/3
i think 0000003f have something interesting too

075b133b6181e058/0000003a--3043e775b6/25

nelsonjchen, 2024-12-04:
I've made a somewhat pinned branch of master-ci that makes deleter log like hell with rotation to somewhere in `/data` verbosely and intend to "daily it":

https://github.com/nelsonjchen/openpilot/commit/6c88bb09df55251c9b55f0a5206e8b4ba1cd9d09

https://github.com/nelsonjchen/openpilot/tree/33993_master-ci

(I'll rebase it onto a newer master-ci when I feel like it.)

It'll be logging onto the SOM's storage, but that's good since it'll be separate from the NVMe drive. 

That, combined with `fallocate -l 400G /data/media/testfile.sparse` to "trigger" it the storage being "full-ish" so it'll be in the persistently deleting state on my 1TB C3 SSD, should help trace the issue when it breaks again. I had to put this weight back on since I cleared some of the the SSD to get it working again ASAP.

Anyway, just mentioning my methodology if anyone else wants to follow along.

adeebshihadeh, 2024-12-05:
Looked into it a bit more, and it's not actually the NVMe. Seems to be related to internet connectivity like @AlexandreSato mentioned.

nelsonjchen, 2024-12-09:
I can't admit I've driven too much but so far, no hits on the issue. Maybe I need to try a full shutdown/power up away from wifi/internets.

nelsonjchen, 2024-12-09:
@AlexandreSato , anybody who can reliably hit this? maybe they can run the hacky logging patch?

AlexandreSato, 2024-12-09:
> @AlexandreSato , anybody who can reliably hit this? maybe they can run the hacky logging patch?

I think the best way would be to use replay, recreating logs in a loop. Lately, I've been connecting to my phone's hotspot to work around the issue.

nelsonjchen, 2024-12-15:
Unable to reproduce it so far. I guess my habits just aren't compatible. Hey, those who can reproduce it consistently, please cherry-pick my commit and share over some logs. 🤞

korhojoa, 2024-12-16:
Should be very easy to reproduce, happens to me 100% of the time if there is no network connection.
I'll try the branch.

inquisitev, 2024-12-30:
```41f4ae6d3b79a9b1``` is my dongle ID.

i see this issue consistently. I have added a menu item that will show my remaining storage from the developer menu. I observed that the storage went to 38 and stayed there for the next couple hours until i stopped again, then it stopped at 21, then at 10. All of the routes that i drove appeared to be updated, but they were not being deleted from the c3. I do not have comma prime and take lots of long drives. 

When i play with the deleter script to make it start deleting some stuff, it seems to try to delete only one route/seg then stops? there were LOTS of routes here. 
[image]

Since that image, i sent an update with a syntax error and for some reason could not ssh back in, some service not started? ip changed? im not sure. i had to reset. I am confident that i will see this again. Ill monitor it closely and improve the logging as i go.

inquisitev, 2024-12-30:
Ive started a thread in the discord: https://discord.com/channels/469524606043160576/1322679694889324616

incognitojam, 2025-01-09:
> it seems to try to delete only one route/seg then stops? there were LOTS of routes here.

Once the available space reaches 10% or 5GB, it will delete 1 or 2 segments to free up space, but it won't delete more than that so that we keep as many logs as possible. It will only delete more once new segments are created (so while driving it will delete one every minute).

korhojoa, 2025-01-12:
[log.zip](https://github.com/user-attachments/files/18388594/log.zip)
Here's `log.txt` from `/data/33993_log/`.
It would be nice to have more information on how much it's filling up so drives wouldn't be blind until it suddenly is too full and stops in a little while.

jyoung8607, 2025-01-12:
It's happening due to a time warp, which can be seen in @korhojoa's logs. The clock is going back to 2015, trigger unknown but isn't supposed to matter. Threading event.wait() is supposed to use a monotonic clock and therefore be robust to that, but it isn't. @incognitojam found python/cpython#91394 which may describe the issue.

To reproduce on device, apply this patch:

```diff
diff --git a/system/loggerd/deleter.py b/system/loggerd/deleter.py
index 2a2bd5fe1..44e796a62 100755
--- a/system/loggerd/deleter.py
+++ b/system/loggerd/deleter.py
@@ -69,7 +69,8 @@ def deleter_thread(exit_event: threading.Event):
           cloudlog.exception(f"issue deleting {delete_path}")
       exit_event.wait(.1)
     else:
-      exit_event.wait(30)
+      cloudlog.info("Sleeping")
+      exit_event.wait(5)
```

Then launch `deleter` in one terminal:

```
comma@comma-2a02b9c8:/data/openpilot$ LOGPRINT=debug system/loggerd/deleter.py  
Sleeping                               
Sleeping                               
Sleeping       
```

Then tamper with the system time in another terminal:

```
sudo systemctl stop systemd-timesyncd
sudo date --set "1 Jan 2015 12:00:00"
date
```

You will stop seeing "Sleeping" output from deleter, but it will resume when the clock is fixed:

```
sudo systemctl start systemd-timesyncd
```

This explains why deleter recovers when we get Internet connectivity, because `timesyncd` can fix the time.

The correct fix is not yet clear to me. There are a number of hacks we could apply, but ideally if Python is supposed to be using a monotonic clock for Event.wait(), we should find out why it isn't and fix that.

jyoung8607, 2025-01-12:
Following up to myself:

The same problem cannot be reproduced on release; the threading wait() call doesn't seem vulnerable to system time changes. It remains to be determined what changed in Python or AGNOS that broke monotonic timekeeping, but having identified that it's a regression, it should be fixable somehow.

**Correct behavior with:**
```
comma@comma-2a02b9c8:/data/openpilot$ python --version
Python 3.11.4
comma@comma-2a02b9c8:/data/openpilot$ cat /VERSION
10.1
```

**Wrong behavior with:**
```
comma@comma-2a02b9c8:/data/openpilot$ python --version
Python 3.12.3
comma@comma-2a02b9c8:/data/openpilot$ cat /VERSION 
11.4
```

adeebshihadeh, 2025-01-12:
Very interesting! Should be easy to remove this use of `threading.Event.wait`, but it's probably good to fully root cause this one...

adeebshihadeh, 2025-01-15:
If this is really a Python bug, not sure there's much we can do here besides block the usage of `threading.Event.wait`. See how we do this for `pytest.main`: https://github.com/commaai/openpilot/blob/2258ea310ddcd7caeb8f63ede3ae4656d0bffe3f/pyproject.toml#L255.

jyoung8607, 2025-01-15:
Python clearly goes to some effort to use a monotonic clock wherever available, but there are also platforms where there isn't one, and it may depend on underlying things like glibc. It's kind of a rabbit hole.

https://bugs.python.org/issue23428
https://bugs.python.org/issue41710

It's definitely possible to fix our Python, it used to work in AGNOS 10, and it works on desktop. The question is whether to spend the time tracing down why only the AGNOS 11 build is busted, or do we take the cheap way out.

If we take the cheap way out, we need to make sure clock rollbacks are included in what runs on the comma test rack. TBH that should probably happen anyway.

adeebshihadeh, 2025-01-18:
AGNOS 11.5 https://github.com/commaai/openpilot/pull/34413 bumps our Python to 3.12.8. 

Unfortunately, it's still not fixed there, and we can't go to 3.13 yet.

It's possible it's just an issue with uv's Python builds (https://github.com/astral-sh/python-build-standalone).

jyoung8607, 2025-01-18:
> It's possible it's just an issue with uv's Python builds (https://github.com/astral-sh/python-build-standalone).

I suspect exactly that, think an upstream build config issue oops like we had with [libzmq](https://github.com/commaai/openpilot/issues/23666#issuecomment-1071002173).

jyoung8607, 2025-01-21:
> It's possible it's just an issue with uv's Python builds (https://github.com/astral-sh/python-build-standalone).

Confirmed.

```python
#!/usr/bin/env python3
import threading

def testing_thread(test_event: threading.Event):
  while True:
    print("Sleeping...")
    test_event.wait(1)

def main():
  testing_thread(threading.Event())

if __name__ == "__main__":
  main()
```

This reproduces the problem on AGNOS 11.4. Switching to system Python (`/usr/bin/python3`) makes it work correctly.

The system Python just happens to be **exactly the same Python version**, built differently.

```
comma@comma-2a02b9c8:/data/openpilot$ python3
Python 3.12.3 (main, Apr 15 2024, 17:09:49) [GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ^D
comma@comma-2a02b9c8:/data/openpilot$ /usr/bin/python3
Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

In the Astral repo I can't find any PRs or issues referencing the clock issue, but [Astral makes **extensive** build and code changes](https://github.com/astral-sh/python-build-standalone/tree/main/cpython-unix). For starters, you can see above the UV build is done with an ancient GCC. It acts differently even in the output above, the Ctrl-D to exit is consistently echoed to screen in the UV build but not echoed in the system build.

It comes with its own threading library (which is right in our problem area):

```
comma@comma-2a02b9c8:/data/openpilot$ ls -al /usr/local/uv/python/cpython-3.12.3-linux-aarch64-gnu/lib
total 48340
drwxr-xr-x 10 root root     4096 Nov 13 18:02 .
drwxr-xr-x  6 root root     4096 Nov 13 18:02 ..
drwxr-xr-x  2 root root     4096 Nov 13 18:02 itcl4.2.2
lrwxrwxrwx  1 root root       20 Nov 13 18:02 libpython3.12.so -> libpython3.12.so.1.0
-rwxrwxr-x  1 root root 49387544 Nov 13 18:02 libpython3.12.so.1.0
-rwxrwxr-x  1 root root    66328 Nov 13 18:02 libpython3.so
drwxr-xr-x  2 root root     4096 Dec 17 00:14 pkgconfig
drwxr-xr-x 38 root root     4096 Dec 17 00:14 python3.12
drwxr-xr-x  5 root root     4096 Nov 13 18:02 tcl8
drwxr-xr-x  6 root root     4096 Nov 13 18:02 tcl8.6
drwxr-xr-x  2 root root     4096 Nov 13 18:02 thread2.8.7
drwxr-xr-x  4 root root     4096 Nov 13 18:02 Tix8.4.3
drwxr-xr-x  6 root root     4096 Nov 13 18:02 tk8.6
```

Shipping both copies of Python 3.12.3 inflates the AGNOS image by almost 100MB:

```
comma@comma-2a02b9c8:/data/openpilot$ du -sh /usr/local/uv/python/cpython-3.12.3-linux-aarch64-gnu
96M     /usr/local/uv/python/cpython-3.12.3-linux-aarch64-gnu
```

Insert do-not-want GIF here.

After the move to Ubuntu 24.04, do we [continue to have good reason](https://github.com/commaai/agnos-builder/blob/2a5a908651e9ef56ae435cea074b935512138d1c/TESTING.md?plain=1#L10) to [not use the system Python](https://github.com/commaai/agnos-builder/blob/2a5a908651e9ef56ae435cea074b935512138d1c/userspace/openpilot_python_dependencies.sh#L12)?

adeebshihadeh, 2025-01-23:
Fixed in https://github.com/commaai/openpilot/pull/34413

closed 2025-01-23 by adeebshihadeh