commaai/openpilot #33996: Comma 3X processes keep failing during drive

spinkickr, 2024-11-12:
### Describe the bug

I recently purcahsed a Comma 3X for my 2022 Chevy Bolt EV and installed it on Sunday. Ran through the initial setup, installed openpilot, etc. I had the unit working and no issues at all. However, on the 2nd drive of having the unit installed, I would try to engage the ACC on the car and would de-engage immediately and get a "Process Not Running: qcomgpsd" error on the Comma 3X screen. No matter how many times you tried to engage the ACC, that error would show up and not allow the car to engage the ACC. To resolve this issue, I would have to go into the settings of the comma unit and restart it for it to work again. I then had an instance where I was on the highway and the whole unit went into panic mode and said something along the lines of "Take control of the car immediately! Unit has stopped an essential process" and would cut the ACC/steering assist to the car not allowing me to re-enable features on the comma with that process error listed above. Happened a few more times, so I re-installed the openpilot software on the device which entails, factory resets the unit. Worked fine for the first drive and today on my way to and from work, but got the error again while getting dinner and now I'm also getting a "Poor GPS reception: Ensure device has a clear view of the sky" error that will not go away. The unit is placed directly in the middle of my windshield away from any obstructions so I know there's a clear path to the satellites. Is this a possible firmware glitch in the unit that would require me to reload the system's firmware onto the unit or possible hardware failure?

### Provide a route where the issue occurs

8939b034486f78ea/00000007--c9f01f2882/0

### openpilot version

0.9.7

### Additional info

Unit will no longer retrieve a GPS signal to the device

adeebshihadeh, 2024-11-12:
Looks like a HW issue. Open a ticket at https://comma.ai/support and include a link to this issue.

closed 2024-11-12 by adeebshihadeh

rozboris, 2025-05-11:
FWIW I seem to be getting the same exact issue (on a completely different car). Opened a new bug report at https://github.com/commaai/openpilot/issues/35173 and contacted support to see if they still think it's hardware.