commaai/openpilot #33998: Forward video has serious tearing artifacts

nelsonjchen, 2024-11-12:
### Describe the bug

The first drive's after a cold boot's forward video is fine. The next ONROAD's forward video is not. Might need the screen to go dark in-between drives. The wide video seems fine on all the drives. Not sure what happens on the third drive.

---

this one's first boot's drive is fine: https://connect.comma.ai/fe18f736cb0d7813/00000404--83516f16c4 

second boot:

https://github.com/user-attachments/assets/e6527965-1691-4b06-8b26-b4adde2ae36a

https://github.com/user-attachments/assets/ffeda102-b76b-4257-affd-5565b44460f1

Context: Vehicle started from cold boot, had a bit of a drive, saw TPMS low tire pressure, went back to house to get tire pump, pumped tire, started second drive, saw errors.

---

first boot is fine: https://connect.comma.ai/fe18f736cb0d7813/0000040a--0dbe63a02e 

https://github.com/user-attachments/assets/d7d51a3f-f911-485d-a100-cde7e910dcfb

https://github.com/user-attachments/assets/94c96e84-9179-430c-b36d-aad36102f124

third to fifth drive seems fine?: 

* https://connect.comma.ai/fe18f736cb0d7813/00000406--f82bf78ec4

* https://connect.comma.ai/fe18f736cb0d7813/00000407--a56e0a2bec

* https://connect.comma.ai/fe18f736cb0d7813/00000408--9eb4d4c06f

Context: Just went out, got something to eat, groceries. 

### Provide a route where the issue occurs

fe18f736cb0d7813/0000040b--78d26612d7/2 , fe18f736cb0d7813/00000405--2b441abe00/9 

### openpilot version

master-ci

### Additional info

Discord links:

* feedback: https://discord.com/channels/469524606043160576/1305837783490891826

* openpilot-experience: https://discord.com/channels/469524606043160576/616456819027607567/1305838954825777162

nelsonjchen, 2024-11-12:
While not yet reported yet, a447729c1d15ff89 [might have the same issue ](https://discord.com/channels/469524606043160576/616456819027607567/1305905832046624768)(patienc3 on Discord). They might come in with a report or data of their own.

choongsoo, 2024-11-13:
Same issue observed in the route https://connect.comma.ai/a447729c1d15ff89/0000001b--0585570655

adeebshihadeh, 2024-11-13:
Thanks for the great report!  We made a bunch of relevant changes here recently (mostly ISP + tinygrad), but nothing that particularly points at the fcam.
* confirmed on both your drive's that it's just fcam (ecam and dcam are fine)
* friom commit 97c990dac6425e6a7eb0299007927ceb6d1c1ea4

adeebshihadeh, 2024-11-13:
Reproduced on an AR231 device. Repro procedure is reboot and cycle ignition a few times.
* on master, 3/4 boot ups had the tearing
* with `BLOCK=modeld,dmonitoringmodeld`, 0/6 boot ups had the issue

MarcoTheDingo, 2024-11-14:
Had the same issue on my trip tonight. Lower third of the forward camera has this issue. Route: 5130484aa8069bad/00000007--b9a36bd4fa

adeebshihadeh, 2024-11-14:
Reverting until this is fixed #34020.

jakethesnake420, 2024-11-30:
I had this happen to me earlier this year. Have not noticed it since. https://connect.comma.ai/164080f7933651c4/2024-04-07--11-04-32

adeebshihadeh, 2024-11-30:
> I had this happen to me earlier this year. Have not noticed it since. [connect.comma.ai/164080f7933651c4/2024-04-07--11-04-32](https://connect.comma.ai/164080f7933651c4/2024-04-07--11-04-32)

This looks a bit different, but may be the same underlying cause. Do you happen to remember whether it looked like that in the UI or just the recorded video?

adeebshihadeh, 2024-12-11:
Alright so more info on this:
* tearing happens once you hit the [output fence timeout](https://github.com/commaai/openpilot/blob/02976db4729207a8600e77f0d9dd6a0fd7b245c2/system/camerad/cameras/spectra.cc#L681)
* once it's timed out, the ISP gets in a bad state and never recovers
* AR is more likely since the readout is 2x longer than OX

closed 2024-12-12 by adeebshihadeh