commaai/openpilot #33999: locationd error after u-turn

YassineYousfi, 2024-11-12:
### Describe the bug

openpilot unavailable - locationd temporary error after a u-turn.

### Provide a route where the issue occurs

09ed4c7e7b4937fb/00000290--440bddbb31/14

### openpilot version

d239c7f3252e2bbe6d94356acbc831ed53e23569

adeebshihadeh, 2024-11-20:
Unclear if it's similar, but another one:
> harald
>   7:14 PM
> https://connect.comma.ai/a39a2735e8e190d3/000002af--32922ff50b
> Had locationd errors all drive

sshane, 2025-01-22:
@fredyshox do you know if we fixed this?

fredyshox, 2025-01-22:
This is something else. Its related to camodo/gyro yaw rate cross check failing (added in: https://github.com/commaai/openpilot/pull/29727), due to highly uncertain camodo yr estimates deviating from gyro measurements during the u-turn. 

[image]

Its not a regression.

adeebshihadeh, 2025-01-22:
Is this alert a FP? What would happen if we didn't aelrt?

PR #34476 (2025-01-25): locationd: frequency based bad observation resiliance and recovery
Fixes #33999
closed 2025-01-25 by fredyshox